microsoft/Azure-Kinect-Sensor-SDK #623: Add function to get bytes per pixel from image format

UnaNancyOwen, 2019-08-09:
I want to add function to get bytes per pixel from image format.
That idea was provide by @Brent-A during this [discussion](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/pull/596#discussion_r311828112).

**Is your feature request related to a problem? Please describe.**

It is useful when specifying stride bytes in <code>k4a_image_create()</code> or <code>k4a::image::create()</code>.
```cpp
// Before
k4a::image output_image = k4a::image::create(source_image.get_format(), 
                                             output_width,
                                             output_height,
                                             output_width * 4 * static_cast<int32_t>(sizeof(uint8_t)));

// After
const int32_t bytes_per_pixel = k4a_get_pixel_size_from_image_format(source_image.get_format());
k4a::image output_image = k4a::image::create(source_image.get_format(), 
                                             output_width,
                                             output_height,
                                             output_width * bytes_per_pixel);
```

**Describe the solution you'd like**

Add function to get bytes per pixel from image format.
```c
K4A_EXPORT int32_t k4a_get_pixel_size_from_image_format(k4a_image_format_t format);
```
```c
int32_t k4a_get_pixel_size_from_image_format(k4a_image_format_t format)
{
    switch (format)
    {
    case K4A_IMAGE_FORMAT_COLOR_BGRA32:
        return (int32_t)sizeof(uint8_t) * 4;
    case K4A_IMAGE_FORMAT_DEPTH16:
    case K4A_IMAGE_FORMAT_IR16:
        return (int32_t)sizeof(uint16_t);
    case K4A_IMAGE_FORMAT_CUSTOM8:
        return (int32_t)sizeof(int8_t);
    case K4A_IMAGE_FORMAT_CUSTOM16:
        return (int32_t)sizeof(int16_t);
    case K4A_IMAGE_FORMAT_CUSTOM:
        return (int32_t)sizeof(int16_t) * 3;
    default:
        return -1;
    }
}
```

**Describe alternatives you've considered**

The function returns bytes per pixel by input image format.

**Additional context**

Brent-A, 2019-08-09:
Here are some example locations where this type of functionality might be useful. It's worth reviewing these to evaluate how well a solution will work for the various callers.

* [rgbz.c](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/c0800a43fb42200935aa9a8da58b8bd5db3d55c9/src/transformation/rgbz.c#L709)
* [image.c](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/2a5027f1d734f4ef86b01520ffc116b016e1d825/src/image/image.c#L178)
* [Image.cs](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/2a5027f1d734f4ef86b01520ffc116b016e1d825/src/csharp/SDK/Image.cs#L73)
* [matroska_read.cpp](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/708987a90b9dcdd178d5c6ef857c3db81e9d2324/src/record/internal/matroska_read.cpp#L781)
* [matroska_write.cpp](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/708987a90b9dcdd178d5c6ef857c3db81e9d2324/src/record/internal/matroska_write.cpp#L72)
* [mfcamerareader.cpp](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/7a9237f88cf39954980f581db0321e25251966e1/src/color/mfcamerareader.cpp#L926)
* [uvc_camerareader.cpp](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/7a9237f88cf39954980f581db0321e25251966e1/src/color/uvc_camerareader.cpp#L1184)
* [k4a.hpp](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/2a5027f1d734f4ef86b01520ffc116b016e1d825/include/k4a/k4a.hpp#L922)

Some trends I see:
1) Most call sights are only interested in a subset of the image formats (e.g. they only really deal in CUSTOM* or in BGRA or they explicitly don't deal with MJPEG).
2) NV12 is a complicated case where "pixel size" doesn't always get the desired result
3) YUY2 is another case where "pixel size" is awkward.

Design considerations:
1) K4A_IMAGE_FORMAT_CUSTOM's pixel size is undefined. It's 6 bytes for the point cloud generation, but that format is left open to interpretation for other arbitrary user data types. It might be better to create a new K4A_IMAGE_FORMAT_POINT_CLOUD to be explicit since we've trended toward more specialized format codes that we originally expected.
2) This is also a bit related to issue #40. We have a few cases where it would be useful to have programmatic metadata about enum types rather than just the documentation. If we add new APIs for getting metadata on enum types they should be consistent.
3) Should pixels sizes be assumed to be multiples of bytes? (e.g. is there a potential for 12 bit pixels) Stride is defined in bytes since it ensures byte alignment of lines, but pixels formats can be somewhat arbitrarily packed. In YUY2 for example, being more akin to *average* pixel size.

Some alternatives to consider:
1) A function that returns minimum stride given a pixel format and a width?
2) modify k4a_image_create to accept 0 as the stride parameter and automatically allocate an appropriate minimum stride for formats that it can be done with?

Brent-A, 2019-08-09:
Related issue #587. This was a case where stride/pixel size/buffer size were being interpreted incorrectly.

UnaNancyOwen, 2019-08-09:
> It might be better to create a new K4A_IMAGE_FORMAT_POINT_CLOUD to be explicit since we've trended toward more specialized format codes that we originally expected.

I agree. I think it make clarify purpose of k4a::image than K4A_IMAGE_FORMAT_CUSTOM.

UnaNancyOwen, 2019-08-09:
> Should pixels sizes be assumed to be multiples of bytes?
> A function that returns minimum stride given a pixel format and a width?

It makes sense. It is fit with this purpose.

UnaNancyOwen, 2019-08-09:
> Here are some example locations where this type of functionality might be useful. It's worth reviewing these to evaluate how well a solution will work for the various callers.

Thank you for your research. The API that provides stride seems useful in many case.

Brent-A, 2019-08-09:
I'm thinking that just modifying k4a_image_create to succeed when you pass in a zero stride might be the simplest. In most cases the stride is being calculated just to satisfy the argument to k4a_image_create anyhow.

PR #819 (2019-10-08): Enabled implicit stride in k4a_image_create

## Fixes #623

### Description of the changes:
- Modified k4a_image_create to allow "0" as a valid stride. When zero is specified, k4a_image_create infers the minimum necessary stride appropriate for the image format. 

### Before submitting a Pull Request:
- [X] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [X] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [X] I ran the [unit tests](https://github.com/Microsoft/Az [... cut]
closed 2019-10-08 by Brent-A