microsoft/Azure-Kinect-Sensor-SDK #625: Azure Kinect Viewer crashes when logging is enabled

rfilkov, 2019-08-09:
**Describe the bug**

Azure Kinect Viewer works fine, if logging is not enabled or if it is started with 'Run as administrator'. But if logging is enabled and sensor is connected (or recording is opened), it crashes without any warning or message. I suppose this is due to insufficient rights to save log-files in the Azure Kinect installation folder.

**To Reproduce**

1. Set K4A_ENABLE_LOG_TO_A_FILE=k4a.log & K4A_LOG_LEVEL=W.
2. Connect Azure Kinect sensor.
3. Start Azure Kinect Viewer

**Expected behavior**

I expect Azure Kinect Viewer to work as expected, no matter if the logging is enabled or not.

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: Windows 1809
 - SDK Version:  1.1.1
 - Firmware version:  factory default

**Additional context**

wes-b, 2019-08-09:
This should be no repro in 1.2.0.

rfilkov, 2019-08-12:
The same applies to the body tracking viewer, as well.

tesych, 2019-08-12:
@rfilkov, I tried your repro steps with SDK 1.2 beta that we just released and I cannot repro the bug. Can you please confirm that you still seeing this issue even when using SDK 1.2? 
Also, can you share your Firmware version?

rfilkov, 2019-08-12:
@tesych Sure. I'll check it, right after 1.2.0 msi is available. Could you please try the body tracking viewer as well, while the BT logging is turned on.

As I said above, my device's firmware version is not changed (yet):
- RGB camera: 1.6.98
- Depth camera: 1.6.70
- Audio: 1.6.14
- Build config: Release

cdedmonds, 2019-08-15:
@yijiew can you take a look at the body tracking viewer?

yijiew, 2019-08-16:
@rfilkov May I ask what is the detailed repro step when you try the body tracking viewer?

rfilkov, 2019-08-16:
@yijiew Of course. Here are the steps:
1. set K4A_ENABLE_LOG_TO_A_FILE=k4a.log & K4A_LOG_LEVEL=I (or W).
2. Run the command prompt (cmd). Then: cd C:\Program Files\Azure Kinect Body Tracking SDK\tools
3. Make sure the sensor is connected. Start k4abt_simple_3d_viewer.exe. It will run for a second or two, and then close without any warning or error message.
4. Close the command prompt window. Remove K4A_ENABLE_LOG_TO_A_FILE or set it to 0.
5. set K4ABT_ENABLE_LOG_TO_A_FILE=k4abt.log & K4ABT_LOG_LEVEL=I (or W).
6. Repeat steps 2 & 3. You will get the same result.
7. Remove both K4A_ENABLE_LOG_TO_A_FILE & K4ABT_ENABLE_LOG_TO_A_FILE or set them to 0.
8. Repeat steps 2 & 3. This time the viewer will show up, as expected.

I suppose the logger tries to create its log-file in this Program Files subfolder. Because it doesn't have sufficient rights to do it, the log-file creation fails, the error is not catched and this causes the viewer to crash. If you run the viewer as administrator or logging is not enabled, there is no problem.

wes-b, 2019-08-19:
> If you run the viewer as administrator or ...
 This is definitely no repro for 1.2.0 in both the ADK and ABT as we detect the failure and printf the error.

closed 2019-08-19 by wes-b

rfilkov, 2019-08-28:
> This is definitely no repro for 1.2.0 in both the ADK and ABT as we detect the failure and printf the error.

Not sure why you've closed this issue in the meanwhile.
The issue is resolved in Azure Kinect Viewer v1.2.0, but still persists in Azure Kinect Body Tracking Viewer v0.9.1.

rfilkov, 2019-08-28:
@yijiew, thank you for the clarification!