microsoft/Azure-Kinect-Sensor-SDK #627: libdepthengine.so.2.0 is missing for linux

bearpaw, 2019-08-09:
**Describe the bug**

`libdepthengine.so.2.0` is missing for linux. While the latest package is [`libk4a1.1`](https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/)

**To Reproduce**

**Expected behavior**

  ```
  Capturing 100 frames
  [2019-08-09 14:32:09.353] [error] [t=2031] ../Azure-Kinect-Sensor-SDK/src/dynlib/dynlib_linux.c (86): dynlib_create(). Failed to load shared object libdepthengine.so.2.0 with error: libdepthengine.so.2.0: cannot open shared object file: No such file or directory
  [2019-08-09 14:32:09.353] [error] [t=2031] ../Azure-Kinect-Sensor-SDK/src/deloader/deloader.c (70): load_depth_engine(). Failed to Load Depth Engine Plugin (depthengine). Depth functionality will not work
  [2019-08-09 14:32:09.353] [error] [t=2031] ../Azure-Kinect-Sensor-SDK/src/deloader/deloader.c (71): load_depth_engine(). Make sure the depth engine plugin is in your loaders path
  [2019-08-09 14:32:09.353] [error] [t=2031] ../Azure-Kinect-Sensor-SDK/src/deloader/deloader.c (110): deloader_depth_engine_create_and_initialize(). Failed to load depth engine plugin
  [2019-08-09 14:32:09.353] [error] [t=2031] ../Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (154): depth_engine_start_helper(). Depth engine create and initialize failed with error code: 108.
  [2019-08-09 14:32:09.353] [error] [t=2031] ../Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (160): deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper()
  [2019-08-09 14:32:09.353] [error] [t=2031] ../Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (194): depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread()
  [2019-08-09 14:32:09.353] [error] [t=2027] ../Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (552): dewrapper_start(). Depth Engine thread failed to start
  [2019-08-09 14:32:09.353] [error] [t=2027] ../Azure-Kinect-Sensor-SDK/src/depth/depth.c (451): dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start()
  [2019-08-09 14:32:09.359] [error] [t=2027] ../Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (359): cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
  [2019-08-09 14:32:09.359] [error] [t=2027] ../Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (362): depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
  [2019-08-09 14:32:09.363] [error] [t=2027] ../Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (891): depth_start(device->depth, config) returned failure in k4a_device_start_cameras()
  Failed to start device
  ```
  
**Logs**

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 16.04 
 - SDK Version:  commit `2a5027f1d734f4ef86b01520ffc116b016e1d825` 
 - Firmware version:  
```
 == Azure Kinect DK Firmware Tool == 
Device Serial Number: 000423592912
Current Firmware Versions:
  RGB camera firmware:      1.6.98
  Depth camera firmware:    1.6.70
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft
```

**Additional context**

wes-b, 2019-08-09:
Hi Bearpaw, you can get the binaries from the NuGet package we have released: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/608.

closed 2019-08-09 by wes-b

jrterven, 2019-08-14:
Hi wes-b, I have the same problem. How do we install the binaries from the NuGet package in Ubuntu?

liaoyinyu, 2019-08-15:
libdepthengine.so.2.0 is not avaliabel for the lasted version. you can checkout to the master branch  and build the sdk. Looking forward to the libdepthengine.so.2.0 support the linux version.

fishcu, 2019-08-23:
@jrterven Were you able to get the necessary files from your NuGet installation?

I'm in the same boat. However, meanwhile, I was able to use the latest SDK with the older depth engine by changing the hard-coded library version:

`#define K4A_PLUGIN_MAJOR_VERSION 1`

in `k4aplugin.h`. This will load `libdepthengine.so.1.0`.

wes-b, 2019-08-23:
Both the 1.2.0 NuGet and MSI have libdepthengine.so.2.0 in it. We are working to also get the Debian package released early next week.

adamconkey, 2019-09-26:
Any update on the Debian release?

RoseFlunder, 2019-09-26:
@adamconkey
On ubuntu 18.04 its available via the Microsoft Package Repository.
You can install sensor sdk 1.2.0 (with libdepthengine.so.2.0) and body tracking sdk 0.9.3 from the repository
https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download#linux-installation-instructions

adamconkey, 2019-09-26:
@RoseFlunder Thank you for the update. I wanted it for Ubuntu 16.04 but was able to work around it by extracting the contents of the MSI file for Windows, and just copied the binaries to the SDK bin folder.

wes-b, 2019-09-26:
> I wanted it for Ubuntu 16.04 but was able to work around it by extracting the contents of the MSI file for Windows, and just copied the binaries to the SDK bin folder.

Unfortunately we are only targeting Ubuntu 18.04 right now. Please feel free to ask for support of more platforms and distributions here: https://aka.ms/azurekinectfeedback.

Gangareddy, 2019-11-13:
I'm in the same boat, my compile is successful for arm. But can't find the shared libary (libdepthengine.so.2.0). I was able to load the libdepthengine.so.2.0 from (https://github.com/ryanmjacobs/AUR-azure-kinect-sensor-sdk-git/commit/0847283a9e098f3a3c5c01e720ae39a1e915e131) and link it using LD_LIBRARY_PATH . But it still complains of missing lidepthengine.so .

JianxiangFENG, 2021-02-07:
> I'm in the same boat, my compile is successful for arm. But can't find the shared libary (libdepthengine.so.2.0). I was able to load the libdepthengine.so.2.0 from ([ryanmjacobs/AUR-azure-kinect-sensor-sdk-git@0847283](https://github.com/ryanmjacobs/AUR-azure-kinect-sensor-sdk-git/commit/0847283a9e098f3a3c5c01e720ae39a1e915e131)) and link it using LD_LIBRARY_PATH . But it still complains of missing lidepthengine.so .
hey,
I got the same error, have you solved it ?

natejenkins, 2021-02-13:
Hey @JianxiangFENG , what worked for me was downloading the library from https://www.nuget.org/api/v2/package/Microsoft.Azure.Kinect.Sensor/1.4.0-alpha.4 and placing libdepthengine.so.2.0 in build/bin.  The instructions here: https://github.com/valdivj/Azure-for-Kinect-Jetson-nano were helpful to me.

DatleUNL, 2023-11-22:
Hi everyone, I'm running pyinstaller in a conda environment, and I also ran into this same issue. I have included the libdepthengine.so.2.0 in every paths I can think of: whereis libdepthengine.so.2.0
libdepthengine.so.2: /lib/x86_64-linux-gnu/libdepthengine.so.2.0 /usr/lib/x86_64-linux-gnu/libdepthengine.so.2.0 /usr/local/lib/libdepthengine.so.2.0

The app still couldn't find the lib. Could someone please help me?