microsoft/Azure-Kinect-Sensor-SDK #629: Zip Download doesn't work

jimjsong, 2019-08-11:
Using the download of the project doesn't work.  It reports missing CMakeLists.txt for multiple submodules.  Wasted a few hours interpreting the error to mean that I need to clone the repo and not just down the zip.

xthexder, 2019-08-12:
The source code zip files are auto-generated by github, so it is not possible to include submodules as part of these zips. To build from source you will need to clone the repository.
If you don't need to build from source, you can download one of the prebuilt releases available on the usage page: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md

jimjsong, 2019-08-12:
I freely admit I'm new to Kinect development.  So, I didn't know what to expect.  How about just warning not to download the zip?  It would have saved me a few hours.

cdedmonds, 2019-08-15:
@jimjsong good suggestion we can add that to the documentation.

brookman, 2019-08-16:
I got stuck a few hours as well because of this. I suggest adding a warning in the "Building" instructions: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md

tesych, 2019-09-03:
added [documentation](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) to clear this out.

closed 2019-09-04 by tesych