microsoft/Azure-Kinect-Sensor-SDK #631: Arch Linux support

ichlubna, 2019-08-12:
Hello,
while Ubuntu belong to the mainstream Linux distros Arch is widely used by programmers too. I guess I am not the only one who'd appreciate an Arch (pacman) package or AUR one. Compiling with CMake seems to be a bit tricky because of various dependencies which might differ on Arch.
Thanks and have a nice day!

xthexder, 2019-08-12:
I've built K4A successfully on an Arch Linux machine at home. I think it's unlikely that we'll officially support Arch Linux, but I can try and get you a list of the equivalent Arch dependencies, which should make building easier.

ryanmjacobs, 2019-08-12:
Also running Arch here -- an AUR package would be amazing.

xthexder, 2019-08-13:
As it turns out, I was mistaken about building K4A on arch linux.
Since Arch Linux provides extremely recent versions of Clang and GCC, some additional warnings are detected in the source that prevent it from building.

I'll spend some time this week going through the warnings and getting K4A to build.

ichlubna, 2019-08-13:
Hey I managed to build it after all (still would appreciate AUR package though). My first mistake was that I didn't clone --recursive submodules. Second thing was that I had to edit the build.ninja file and replace all -Werror flags with -Wno-error. The examples seem to be working now despite the warnings.
Also had to get the depthengine 2.0 library from the latest NuGet package cause the Debian one had 1.0.

breiting, 2019-08-19:
I just built the source code on my Arch machine. Besides some minor fixes, the built went well. I will receive the hardware tomorrow (being excited about it). Here are the steps which were necessary to build on my Arch system:

## Prerequisite

Make sure that you have installed the [libsoundio](https://github.com/andrewrk/libsoundio) (which is not available in the default Arch repo). I just cloned it and performed a `make && make install`.

## Patch source code

Since the default libsoundio headers are installed under `/usr/local/soundio`, I needed to patch one file:

```diff --git a/tools/k4aviewer/k4asoundio_util.h b/tools/k4aviewer/k4asoundio_util.h
index c9c2718..011a76e 100644
--- a/tools/k4aviewer/k4asoundio_util.h
+++ b/tools/k4aviewer/k4asoundio_util.h
@@ -16,7 +16,7 @@
 // This disables that behavior.
 //
 #define NOMINMAX
-#include <soundio.h>
+#include <soundio/soundio.h>

 // Project headers
 //
```

After that, everything compiled and linked like a charm:

```
mkdir build
cd build
cmake ..
```

Please make sure to update the LD_LIBRARY_PATH before executing any binary. This is required since libsoundio installs the binaries under `/usr/local/lib64`

```
LD_LIBRARY_PATH=/usr/local/lib64 ./bin/k4aviewer
```

xthexder, 2019-08-19:
I was able to build without any modification by installing the `libsoundio` AUR package.

ryanmjacobs, 2019-08-20:
All right, well I took a stab at creating the AUR package (https://github.com/ryanmjacobs/AUR-azure-kinect-sensor-sdk-git). Everything builds and installs fine on my test Arch Linux VM.

This works great for me:
```console
$ yay -S azure-kinect-sensor-sdk-git
```

---

I can access the color camera with `k4aviewer`. However, the depth functionality is broken. I don't have `libdepthengine` anywhere on my system. Not sure where to grab that.

I also had to create a udev rule in order to access the camera:
```
# /etc/udev/rules.d/50-kinect-azure.rules
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", TAG+="uaccess"
```

breiting, 2019-08-20:
According to `docs/building.md` the depth engine is closed source and only available as binary. I will try to get it out from the Ubuntu package and see if it working.

From building.md:

* [Depth Engine](depthengine.md). The depth engine (DE) is a closed source binary shipped with the SDK installer. The DE binary needs to be copied to the host system and added to the path. **NOTE:** *This step is not need for building, but is required to run the SDK and tests*

breiting, 2019-08-20:
Finally, after some fiddling I am able to access the depth camera. However, the path is not straight forward, I had to install `mono` and `nuget` (in a virtual machine, not to pollute my machine). You have to get the latest beta of v1.2.0 which includes the libdepthengine.so.2 file. After having the package installed, I copied over the .so.2 file to my Arch machine and voila.

The distribution of the (closed source) Linux binary is not nice via Nuget.

This command downloads the latest Azure binaries via nuget under Ubuntu 18.04
```console
nuget install Microsoft.Azure.Kinect.Sensor -Version 1.2.0-beta.1
```

xthexder, 2019-08-20:
A couple notes:

We actually provide a udev rule in the git repo that is specific to Kinect: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/scripts/99-k4a.rules

Normally the depth engine is provided as part of the debian package, but the current develop branch references the 2.0 version which so far is only available in beta releases.
I'll have a talk with the team to see what we can do to make depth engine binaries more available.

ryanmjacobs, 2019-08-21:
Great, thanks for the info. I've updated the AUR package to use the udev rules provided in the repository. The thing that confused me was all of the different USB Product IDs -- why do they change each time I remove and reinsert the device? It seems to cycle through `097a` and `097e`.

I've also followed @breiting's advice and used `nuget` to grab a beta release that included `libdepthengine.so.2.0`. I included the binary in my AUR package and everything works great! I'm able to pan around in the 3D view mode and it looks fantastic.

Gangareddy, 2019-11-07:
I have tried the following but no luck on mu build process. 

```mkdir build && cd build && cmake -GNinja -DCMAKE_BUILD_TYPE=Debug -DCMAKE_C_COMPILER=nvcc -DCMAKE-CXX_COMPILER=g++ -debug-output ..```

The output after ten mins is 

``` Running with debug output on. ```

This seems wrong as I can't see any info and the process seems idle. Any idea for building on aarch64 (jetson tx1) ?

FYI: The CMake on aarch64 is very old and I had to build and install from source code.

Gangareddy, 2019-11-07:
> Hey I managed to build it after all (still would appreciate AUR package though). My first mistake was that I didn't clone `--recursive `submodules. Second thing was that I had to edit the [...]

Could you explain the process you followed to build. ?