microsoft/Azure-Kinect-Sensor-SDK #638: hello, how to read and record the infrared video or image like depth (16 bit single channel), infrared (16 bit single channel), rgb (8 bit 3 channels)

zcl912, 2019-08-13:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

rabbitdaxi, 2019-08-14:
@zcl912 is this page [page](https://docs.microsoft.com/en-us/azure/Kinect-dk/record-playback-api) in the documentation website can help resolve your issue?

cdedmonds, 2019-08-15:
@zcl912 please let us know whether that page helps. We will need more details on what you are trying to accomplish to be able to diagnose this correctly.

zcl912, 2019-08-18:
> @zcl912 please let us know whether that page helps. We will need more details on what you are trying to accomplish to be able to diagnose this correctly.

sorry for late reply, the prob, the problem is not solved

rabbitdaxi, 2019-08-19:
@zcl912 if the SDK record/playback API did not solve your problem, can your provide more details of what is your problem exactly is?

wes-b, 2019-08-23:
Hi @zcl912 if you are still having an issue please use our issue template to provide us with more detailed information about the problem. Right now we don't have enough information to help.

>Describe the bug
To Reproduce
Expected behavior
Logs
Screenshots
Desktop (please complete the following information):
OS with Version:
SDK Version:
Firmware version:
Additional context

closed 2019-08-23 by wes-b