microsoft/Azure-Kinect-Sensor-SDK #639: Body tracking crashes due to driver error or dll not found

venscn, 2019-08-13:
**Describe**
When I don't install the driver or lost some dll and to use the body tracking, it crash, nothing error info show me.

**Feature**
First start up and checking driver and dll or provide a function like k4abt_set_error_callback.

cdedmonds, 2019-08-14:
@yijiew can you comment here on the types of missing dependencies which emit log messages in 0.9.1 and which result in crashes?

@venscn this is an issue we are working on. Some issues (such as an out of date driver) do not have a good way for us to produce an error message since the crash is internal to the GPU libraries.

yijiew, 2019-08-14:
When CUDA dlls or Visual C++ Redistribution dlls are missing, the tracker creation will fail and error messages will be logged. But when the driver version is too low to support CUDA 10.0, there isn't a good way to log the error message as @cdedmonds mentioned. We will see what we can do to check the driver version.

qm13, 2019-12-07:
Fixed in v0.9.5.

closed 2019-12-07 by qm13