microsoft/Azure-Kinect-Sensor-SDK #640: Add function to get default configuration for k4a_device_configuration_t

UnaNancyOwen, 2019-08-13:
I propose add function to get default configuration for <code>k4a_device_configuration_t</code>.
Currently, It is necessary to set all <code>k4a_device_configuration_t</code> members in user code.
If default configuration can be retrieved, developer just have only to rewrite necessary member variables.

I think it is developer helpfull. I think it is meaningful to provide this function in SDK.
```cpp
k4a_device_configuration_t k4a_get_default_configuration()
{
    k4a_device_configuration_t configuration;
    configuration.camera_fps                        = K4A_FRAMES_PER_SECOND_30;
    configuration.color_format                      = K4A_IMAGE_FORMAT_COLOR_BGRA32;
    configuration.color_resolution                  = K4A_COLOR_RESOLUTION_1080P;
    configuration.depth_mode                        = K4A_DEPTH_MODE_NFOV_UNBINNED;
    configuration.wired_sync_mode                   = K4A_WIRED_SYNC_MODE_STANDALONE;
    configuration.depth_delay_off_color_usec        = 0;
    configuration.subordinate_delay_off_master_usec = 0;
    configuration.synchronized_images_only          = true;
    configuration.disable_streaming_indicator       = false;
    return configuration;
}
```
```cpp
// Before
k4a_device_configuration_t configuration;
configuration.camera_fps                        = K4A_FRAMES_PER_SECOND_30;
configuration.color_format                      = K4A_IMAGE_FORMAT_COLOR_BGRA32;
configuration.color_resolution                  = K4A_COLOR_RESOLUTION_720P;
configuration.depth_mode                        = K4A_DEPTH_MODE_NFOV_UNBINNED;
configuration.wired_sync_mode                   = K4A_WIRED_SYNC_MODE_STANDALONE;
configuration.depth_delay_off_color_usec        = 0;
configuration.subordinate_delay_off_master_usec = 0;
configuration.synchronized_images_only          = true;
configuration.disable_streaming_indicator       = false;
device.start_cameras( &configuration );

// After
k4a_device_configuration_t configuration = k4a_get_default_configuration();
configuration.color_resolution = K4A_COLOR_RESOLUTION_720P;
device.start_cameras( &configuration );
```

What do you think? Thanks,

xthexder, 2019-08-14:
We currently have a default config initialization called `K4A_DEVICE_CONFIG_INIT_DISABLE_ALL`, which has all the cameras disabled by default.
This shortens the above a little bit to:
```
        k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
        config.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;
        config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
        config.color_resolution = K4A_COLOR_RESOLUTION_1080P;
        config.synchronized_images_only = true;
```

It might make sense to add another helper with some more reasonable defaults, such as the above config.

UnaNancyOwen, 2019-08-14:
@xthexder I updated my pull request. Please see it. Thanks,

wes-b, 2019-08-19:
Sorry for the delay, just back from vacation. I am not supper comfortable calling this 'default' when it is very specific to 720P, NFOV Unbinned, and BGRA32. While it might be good for 1 user, another would still need the code below, which is not much different from the code @xthexder shared. 
```
k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DEFAULT;
config.color_format = K4A_IMAGE_FORMAT_COLOR_MJPG;
config.depth_mode = K4A_DEPTH_MODE_WFOV_UNBINNED;
config.color_resolution = K4A_COLOR_RESOLUTION_1080P;
```
I would be more in favor of a function/macro that took these three values in as parameters.

UnaNancyOwen, 2019-08-23:
@wes-b I see, It can be easily by initialize config with K4A_DEVICE_CONFIG_INIT_DEFAULT.
And, I think your opinion that it is enough is right. (However, It still needs to be setting to more 4-elements until to standard settings of k4aviewer. The difference between me and your opinion is whether it feels troublesome.)
I will withdraw this proposal once. Thanks,

closed 2019-08-23 by UnaNancyOwen