microsoft/Azure-Kinect-Sensor-SDK #641: Docs improvement

jbrownkramer, 2019-08-13:
**Is your feature request related to a problem? Please describe.**

In my dev environment, the build instructions don't work from powershell or a normal cmd prompt.  They do work from the Developer Command Prompt for VS.  It would have saved me an hour or two if that had been stated in the docs.

schultetwin1, 2019-08-13:
Hi @jbrownkramer, thanks for the feedback. In building.md under [Windows Dependencies](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md#windows-dependencies) we do mention that you need to work from a developer command prompt.

>If you are building from a command prompt, it must be a x64 Visual Studio developer command prompt in order for CMake to find the installed compilers. We build both 32-bit and 64-bit binaries, but 64-bit binaries are the only binaries that are tested. (The command prompt should be called something like x64 Native Tools Command Prompt for VS 2017). Note: call the command line tool with the option -arch=amd64 for x64 builds i.e VsDevCmd.bat -arch=amd64

I'll make a change to **bold** the Visual Studio developer command prompt part to make it stand out a little more

schultetwin1, 2019-08-13:
Also, if there is a different spot in our documentation where you believe the Visual Studio Developer Command prompt should be mentioned please feel free to let us know.

PR #642 (2019-08-13): Bolds visual studio dev cmd prompt note
## Fixes #641 

### Description of the changes:
- Bolds instructions on using Visual Studio Developer Command Prompt

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [ ] I ran the [unit tests](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/testing.md)
- [ ] I ran the [functional tests](https://github.com/M [... cut]
closed 2019-08-13 by schultetwin1