microsoft/Azure-Kinect-Sensor-SDK #646: /usr/bin/ld: cannot find -lk4a::k4a

liaoyinyu, 2019-08-15:
**Describe the bug**

I have installed k4atools k4a1.1 and k4a1.1-dev in my ubuntu os.
when i cmake my the example project ,eg fastpointcloud , 
`mkdir build `
`cmake ..`
`make `

Then it showed the error 
`/usr/bin/ld: cannot find -lk4a::k4a
collect2: error: ld returned 1 exit status
CMakeFiles/fastpointcloud.dir/build.make:94: recipe for target 'fastpointcloud' failed
make[2]: *** [fastpointcloud] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/fastpointcloud.dir/all' failed
make[1]: *** [CMakeFiles/fastpointcloud.dir/all] Error 2
Makefile:83: recipe for target 'all' failed`

**Expected behavior**

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version:   ubuntu 18.04
 - SDK Version:   sdk Version 1.1.0

**Additional context**

schultetwin1, 2019-08-15:
Hi liaoinyu, thanks for the bug report. Are you getting this error when you try to build this repository or your own repository?

If its your own repository, can you share your code with us? At least the CMakeLists.txt file?

liaoyinyu, 2019-08-16:
Hi, schultetwin1, My code was modified from the example of fastpointcloud. And the CMakeList.txt file is almost unchanged. 
`cmake `command is correctly work. But when I run  `make` command, the error occured 
`/usr/bin/ld: cannot find  -lk4a::k4a`

schultetwin1, 2019-08-16:
Gotcha, so you’re in your own repository it have copy and pasted the fastcloudpoint example.

Does your CMakeLists.txt file have a call to `find_package(k4a)`?

Otherwise k4a::k4a will not be defined.

liaoyinyu, 2019-08-16:
ok, it works

schultetwin1, 2019-08-16:
Sweet! Not a problem, we are working on getting example CMake projects up.

closed 2019-08-16 by schultetwin1

ZYJxxx, 2019-09-24:
> ok, it works , thanks very much

can you show your cmakelists ?