microsoft/Azure-Kinect-Sensor-SDK #647: Exposure time incorrect in SDK 1.2.0 beta rc

hector-sab, 2019-08-15:
**Describe the bug**
When manually setting the exposure value using k4a_device_set_color_control, the k4a_image_get_exposure_usec returns a different value. The table below show the values returned in my system.

| ETV Manually Set  |  ETV Returned |
| ------------- | ------------- |
| 500  | 500  |
| 1250  | 1250  |
| 2500  | 2500  |
| 8330  | 16670  |
| 16670  | 33330  |
| 33330  | 33330  |
| 41670  | 33330  |
| 50000  | 33330  |
| 66670  | 33330  |
| 83330  | 33330  |
| 100000  | 33330  |
| 116670  | 33330  |
| 133330  | 33330  |

For other system where we have built the SDK 1.2.0, the ETV of 8330 gets 20000.

**To Reproduce**

**Expected behavior**
I would expect the value returned by k4a_image_get_exposure_usec to be the same or close to the one I manually set before.

**Desktop :**
 - OS with Version:Ubuntu 18.04.3 LTS
 - SDK Version: 1.2.0-beta.1-rc
 - Firmware version: 1.6.102

spschul, 2019-08-16:
I haven't gotten anything definitive, but it looks like in `src/color/mfcamerareader.cpp:849` the exposure time gets converted to terms of log seconds. I'm worried that the value is rounded down as part of conversion to a LONG, because it would explain why we're off by a factor of 2.

As for the exposure times getting clipped to 33330, that isn't as surprising. If you're running at 30 FPS, that's roughly the number of microseconds per frame, so you can't really get a higher frame rate. Maybe trying to do so should be an error, though.

wqs111000, 2019-08-17:
Some information in [this website ](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/pull/161)may help. 

As the figure in [this website](https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification#rgb-camera-exposure-time-values) below shows, the RGB camera exposure time has several discrete values. But as far as I  tested, similar to the questioner, parts of  the result doesn't correspond to what's shown in the table.
[image]

Besides, I wonder if the range from 2500us to 10000us is too big. Is it possible to get a intermediate value like 5000 us for a 50hz power freq?

wes-b, 2019-08-19:
@hector-sab I see a couple things going on.
1) I don't know why your results show 8,330 us missing. I would double check your code. We have functional tests running to confirm table @wqs111000 shared: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/9afcd78957615d7d53771148b059c0f8dc53986b/tests/ColorTests/FunctionalTest/color_ft.cpp#L767 One thing our test is doing different then yours is that it runs when the camera is stopped. So that may be a difference. I can't test that today unfortunately as I am WFH.
2) The exposure time will be capped based on the frame rate setting. In the case of the table above its capped for 30FPS.
3) You sometimes see 10,000 us because of https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/448

wes-b, 2019-08-21:
Following up on this. I did test with camera's running and I can not repro your results

closed 2019-08-21 by wes-b