microsoft/Azure-Kinect-Sensor-SDK #648: C# Wrapper - Please Add AnyCPU Support

ve4cib, 2019-08-15:
The current NuGet package explicitly does not support AnyCPU; when trying to compile a project targeting AnyCPU you get an error.

For those of us who have applications that need to target AnyCPU because of other libraries, this is a real bummer.

If we can't get instant AnyCPU support, something like CefSharp's work-around would be greatly appreciated (see: https://github.com/cefsharp/CefSharp/issues/1714).  Short version is that you can edit the .csproj file to disable the error message, and then it's on you to add a handler to AppDomain.CurrentDomain.AssemblyResolve to dynamically set the path to the 32- or 64-bit libraries as necessary.

MsDerekMa, 2019-08-19:
Thanks for the feedback.  We don't currently have plans to directly support AnyCPU because the C# is just a wrapper for the native binaries. We were talking about ways to make it easier to consume if your projects platform doesn't match x86 or x64.

ve4cib, 2019-08-19:
I figured that was probably the case.

In the end what I did was to make a throw-away project that used the NuGet packlage and build it for 32- and 64-bit targets.  Then I copied the requisite DLLs from the bin/{x86|x64}/Debug folder into a content folder in my main application and add a handler to the [...].AssemblyResolve event to set the path to the 64- or 32-bit libraries.  This let me build my application with an AnyCPU target, but it's definitely a little tedious to have to do all of this manually.

closed 2019-09-04 by tesych