microsoft/Azure-Kinect-Sensor-SDK #652: Can't read data from .mkv file

opalmu, 2019-08-16:
I have saved the .mkv file using k4areviewer and currently trying to read data from .mkv file. I followed the link from [this page](https://docs.microsoft.com/en-us/azure/Kinect-dk/record-playback-api#open-a-record-file). After I compiled the code I got the following error:
```
prog.c: In function ‘main’:
prog.c:5:2: error: unknown type name ‘k4a_playback_t’; did you mean ‘k4a_float3_t’?
  k4a_playback_t playback_handle = NULL;
  ^~~~~~~~~~~~~~
  k4a_float3_t
prog.c:5:35: warning: initialization makes integer from pointer without a cast [-Wint-conversion]
  k4a_playback_t playback_handle = NULL;
                                   ^~~~
prog.c:7:6: warning: implicit declaration of function ‘k4a_playback_open’; did you mean ‘k4a_device_open’? [-Wimplicit-function-declaration]
  if (k4a_playback_open(filePath, &playback_handle) != K4A_RESULT_SUCCEEDED)
      ^~~~~~~~~~~~~~~~~
      k4a_device_open
prog.c:13:30: warning: implicit declaration of function ‘k4a_playback_get_last_timestamp_usec’; did you mean ‘k4a_image_set_timestamp_usec’? [-Wimplicit-function-declaration]
  uint64_t recording_length = k4a_playback_get_last_timestamp_usec(playback_handle);
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                              k4a_image_set_timestamp_usec
prog.c:14:26: warning: format ‘%lld’ expects argument of type ‘long long int’, but argument 2 has type ‘uint64_t {aka long unsigned int}’ [-Wformat=]
  printf("Recording is %lld seconds long\n", recording_length / 1000000);
                       ~~~^                  ~~~~~~~~~~~~~~~~~~~~~~~~~~
                       %ld
prog.c:16:2: warning: implicit declaration of function ‘k4a_playback_close’; did you mean ‘k4a_device_close’? [-Wimplicit-function-declaration]
  k4a_playback_close(playback_handle);
  ^~~~~~~~~~~~~~~~~~
  k4a_device_close

```
And the following is the code I am using:
```
#include<stdio.h>
#include<k4a/k4a.h>
int main()
{
	k4a_playback_t playback_handle = NULL;
	const char* filePath = "/home/ryan/C_program/output.mkv";
	if (k4a_playback_open(filePath, &playback_handle) != K4A_RESULT_SUCCEEDED)
	{
	    printf("Failed to open recording\n");
	    return 1;
	}

	uint64_t recording_length = k4a_playback_get_last_timestamp_usec(playback_handle);
	printf("Recording is %lld seconds long\n", recording_length / 1000000);

	k4a_playback_close(playback_handle);
	
	printf("Hello"); 
	return 0;
}
```
The structure of my files:
```
C_program/
  /output.mkv
  /prog.c
```
How I tested the code in terminal:
```
gcc prog.c -o prog
```
I tried to modify the function name and variables as the messages suggested in the terminal but didn't work. How could I fix those issues?

RoseFlunder, 2019-08-17:
You are missing the necessary include.
Try to add this one:
#include <k4arecord/playback.h>

opalmu, 2019-08-17:
@RoseFlunder Thanks for your help. I have tried to include that header file and it showed me some new error:
```
/tmp/ccN247xB.o: In function `main':
prog.c:(.text+0x39): undefined reference to `k4a_playback_open'
prog.c:(.text+0x5c): undefined reference to `k4a_playback_get_last_timestamp_usec'
prog.c:(.text+0x98): undefined reference to `k4a_playback_close'
collect2: error: ld returned 1 exit status
```
Do you know how I could fix those issues? Thank you.

closed 2019-08-19 by opalmu

yongming-qin, 2020-04-02:
After trying, I succeeded with `gcc practice.c -I /usr/include -lk4a -lk4arecord`
The dynamic library /usr/lib/x86_64-linux-gnu/libk4arecord.so is also needed.