microsoft/Azure-Kinect-Sensor-SDK #653: Undefined reference to `k4a_playback_get_next_capture' when reading data from recorded.mkv file

opalmu, 2019-08-17:
I am trying to read images from the .mkv file and testing with the code provided in this [example](https://docs.microsoft.com/en-us/azure/Kinect-dk/record-playback-api#read-captures), but I received the following error:
```
/tmp/ccbym4nv.o: In function `main':
program2.c:(.text+0x3f): undefined reference to `k4a_playback_get_next_capture'
program2.c:(.text+0x54): undefined reference to `k4a_capture_release'
collect2: error: ld returned 1 exit status
```
Below is the code I am using:
```
#include<stdio.h>
#include<k4a/k4a.h>
#include <k4arecord/k4arecord_export.h>
#include <k4arecord/playback.h>
#include <k4arecord/record.h>

int main()
{
	k4a_playback_t playback_handle = NULL;
	k4a_capture_t capture = NULL;
	k4a_stream_result_t result = K4A_STREAM_RESULT_SUCCEEDED;
	while (result == K4A_STREAM_RESULT_SUCCEEDED)
	{
	    result = k4a_playback_get_next_capture(playback_handle, &capture);
	    if (result == K4A_STREAM_RESULT_SUCCEEDED)
	    {
		// Process capture here
		k4a_capture_release(capture);
	    }
	    else if (result == K4A_STREAM_RESULT_EOF)
	    {
		// End of file reached
		break;
	    }
	}
	if (result == K4A_STREAM_RESULT_FAILED)
	{
	    printf("Failed to read entire recording\n");
	    return 1;
	}

	printf("Hello");
	return 0;
}
```
My current files structure:
```
C_program/
  /output.mkv
  /program2.c
```
And the command I am using to compile the file:
```
gcc program2.c -o program2
```
How could I fix this issue? Thanks for your help.

RoseFlunder, 2019-08-17:
If you are running gcc manually from the command line you have to specify the library path and library file with the -L and -l options. You may also need to specify an Include path so that gcc finds the header files.
I recommend you read some tutorials how to link a library when compiling with gcc :)

opalmu, 2019-08-19:
Hi @RoseFlunder, thank you so much for your suggestion and sorry about this trivial stuff. I was used to code in python and pretty new to C. I found the header file ```k4a/playback.h``` in ```/usr/include/``` folder and have linked ```k4a/playback.h``` while I was compiling my own program ```program2.c``` with the command
```
gcc -o prog -I/usr/include -Wall program2.c
```
However it still couldn't find those functions. I have checked that there is no corresponding ```.c``` files in the folder ```/usr/include``` but only the function declaration in ```.h``` file. Is there any piece I am missing and do you know how could I fix it? Many thanks!

RoseFlunder, 2019-08-19:
@opalmu
You have to specify the libraries with the "-l" (lowercase L, not uppercase i) option.
Try to compile your program with:
`gcc program2.c -Wall -lk4a -lk4arecord -o prog`

I compiled this small example successfully like that:
```
#include <stdio.h>
#include <k4a/k4a.h>
#include <k4arecord/playback.h>

int main()
{
	k4a_playback_t playback_handle = NULL;
	if (k4a_playback_open("output.mkv", &playback_handle) != K4A_RESULT_SUCCEEDED)
	{
	    printf("Failed to open recording\n");
	    return 1;
	}

	uint64_t recording_length = k4a_playback_get_last_timestamp_usec(playback_handle);
	printf("Recording is %lu seconds long\n", recording_length / 1000000);
	
	k4a_playback_close(playback_handle);
	return 0;
}
```
Placed a corresponding "output.mkv" next to the executable, run the program and got the appropriate output like: Recording is 5 seconds long.
Hope this helps you as a start.

opalmu, 2019-08-19:
It works! Thank you so much for your help!!

closed 2019-08-19 by opalmu