microsoft/Azure-Kinect-Sensor-SDK #654: More information about the orientation quaternion of the joint

wqs111000, 2019-08-17:
The  Structure to define a single joint including an orientation quaternion. I think more information about it are needed. Or is there any detailed description available? Please help me figure out.
```
/** Structure to define a single joint
 * The position and orientation together defines the coordinate system for the given joint. 
 * They are defined relative to the sensor global coordinate system.*/
typedef struct _k4abt_joint_t
{
    k4a_float3_t     position;    /**< The position of the joint specified in millimeters*/
    k4a_quaternion_t orientation; /**< The orientation of the joint specified in normalized quaternion*/
} k4abt_joint_t;
```

RoseFlunder, 2019-08-18:
Maybe this is what you are searching for?
[https://docs.microsoft.com/en-us/azure/kinect-dk/body-joints](https://docs.microsoft.com/en-us/azure/kinect-dk/body-joints)

wqs111000, 2019-08-19:
> Maybe this is what you are searching for?
> https://docs.microsoft.com/en-us/azure/kinect-dk/body-joints

Thanks for your reply. I knew this website which have been read  carefully. "All joint coordinate systems are absolute coordinate systems relative to the depth camera 3D coordinate system." I think the orientation quaternions of a joint is not relative to the one of its parent joint.

As far as I tried,  given a  T-pose body, like the figure below shows, when rotating  from palm down to palm up,  joint poisitions ware almost unchanged and  wrist joints should have obvious changes. But, wrist joints were almost unchanged either in fact. Is it body tracker not good enough to get reliable  orientation quaternions? Or I misunderstood? 
[image]

Beside of wrist joints, the body  estimated by Kinect 2  have joints of fingers. So we can use  them to estimate palm orientation. But now I am confused by the orientation quaternions.

closed 2019-08-30 by qm13

Chris45215, 2019-09-19:
Forgive me for asking for clarity when you've already answered - but does this mean that the skeleton output assumes that the hands are always straight out from the wrist and forearm, as though the detected person is wearing a cast or brace? Or is it capable of determining a position of the palm (so it can detect if a hand is bent far forwards or backwards), even if it does not give an orientation? I understand that orientation may not be possible at the moment if the system cannot determine whether it is seeing the front of back of the palm, but for my uses I only care where the hand is, and not which direction it is facing.

abalanmsft, 2019-09-20:
In the current release it is the former, as if all the hand joints are rigidly attached to the forearm, but we are working on enabling dynamic hand joints soon.