microsoft/Azure-Kinect-Sensor-SDK #655: Add version define for Azure Kinect Body Tracking SDK

UnaNancyOwen, 2019-08-17:
**Is your feature request related to a problem? Please describe.**

Azure Kinect Sensor SDK already has a version definitions (include/k4a/k4aversion.h).
But, Azure Kinect Body Tracking SDK doesn't have a version definitions.
I think it should be added. What do you think?

**Describe the solution you'd like**

Add version definitions for Azure Kinect Body Tracking SDK.
```cpp
#ifndef K4ABTVERSION_H
#define K4ABTVERSION_H

#define K4ABT_VERSION_MAJOR          0
#define K4ABT_VERSION_MINOR          9
#define K4ABT_VERSION_PATCH          1
#define K4ABT_VERSION_PRERELEASE     "public"
#define K4ABT_VERSION_BUILD_METADATA ""

#define K4ABT_VERSION_STR            "0.9.1-public"

#endif
```

**Describe alternatives you've considered**

Add version header file (<code>k4abtversion.h</code>) to Azure Kinect Body Tracking SDK.

**Additional context**

Brekel, 2019-08-18:
Agree that would be very helpful!
Especially now during preview-release cycles, to check and keep our customers on the latest release.

UnaNancyOwen, 2019-08-30:
@yijiew Thanks, When do you think that will be release?

yijiew, 2019-08-30:
@UnaNancyOwen it will be today :D You can download this now at: [msi](https://www.microsoft.com/en-us/download/details.aspx?id=100128) [nuget](https://www.nuget.org/packages/Microsoft.Azure.Kinect.BodyTracking/0.9.2) or apt-get on Linux.

closed 2019-08-30 by yijiew