microsoft/Azure-Kinect-Sensor-SDK #656: Debian Package for Raspberry Pi

hmexx, 2019-08-18:
Would one of the Debian packages available work as-is on a Raspberry Pi, and if not could you provide one?

Thanks

```
uname -a
Linux raspberrypi 4.19.58-v7l+ #1245 SMP Fri Jul 12 17:31:45 BST 2019 armv7l GNU/Linux
```

xthexder, 2019-08-20:
ARM is currently not supported by the depth engine portion of the SDK.
You can see some of the existing discussion here: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/329 / https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/568

Based on the architecture (armv7l), I assume you're using a Raspberry Pi 3, which does not meet the minimum requirements for Kinect (notably Kinect requires USB 3).

closed 2019-08-20 by tesych

hmexx, 2019-08-20:
Thank you for the response @xthexder 

It's actually a Raspberry Pi 4, but I guess the rest of your comment still applies. I'll check out those threads.

argosvr, 2019-11-27:
no news about a support for Pi 4 ?

wes-b, 2019-12-02:
> no news about a support for Pi 4 ?

It doesn't meet our GPU requirements