microsoft/Azure-Kinect-Sensor-SDK #657: is cuda added to the body-tracking-samples

dfamil, 2019-08-18:
IN body tracking simple 3d viewer.exe it shows cuda loading. but the version on github it does not

**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

yijiew, 2019-08-20:
Yes. The CUDA binaries are shipped in the MSI package. While the sample on GitHub only depends on the body tracking NuGet package, it doesn't have CUDA dlls there. In the future release, we are planning to create a body tracking dependency package which will solve this issue.

lesterlo, 2019-08-21:
Seems yes, the CUDA binaries are embedded in the Windows and Linux Azure Sensor Body tracking SDK package on v0.9.1. You just need to install an Nvidia GPU and proper GPU driver. 

I tested on my ubuntu 18.04 machine. You don't need to install CUDA and cudnn individually to run the body tracking SDK.

closed 2019-08-30 by qm13