microsoft/Azure-Kinect-Sensor-SDK #658: The link of "\extern\libyuv\src" is missed

KaiGuo-Vision, 2019-08-19:
Hello,

I try to download the source code of "\extern\libyuv\src" and failed, because there is no hyperlink for its external source "src @ 9a07219", may I ask your help to add it?  

Best regards,

RoseFlunder, 2019-08-19:
Did you download the source as zip file?
If so, that could be problem.
If you want to build the sdk yourself from source you have to clone the repository with git.

KaiGuo-Vision, 2019-08-19:
Hi,
Right, I downloaded the zip file from internet explorer instead of using git clone, the reason is the git clone is blocked by our company security.

To address this, I searched the Github and find "https://github.com/lemenkov/libyuv". Download and unzip it into \extern\libyuv\src\

I am not sure it is right, please correct me or share me the link of libyuv source code if possible.

Thanks very much.

RoseFlunder, 2019-08-19:
I never tried to download all extern repositories manually but here is how the folder sturcture looks when using git clone:

[image]

So if you place the content of the repository in a "src" folder I guess it could work. But you may have to do it for the other extern repositories as well.

KaiGuo-Vision, 2019-08-19:
Thanks very much.
The folder structure is exactly same as https://github.com/lemenkov/libyuv
And after manually download all of the external repositories, the cmake compiling succeeds.

schultetwin1, 2019-08-19:
Thank you @RoseFlunder for helping out here! You nailed the issue and it sounds like @KaiGuo-Vision is now able to build.

This is an unfortunate side effect of using git submodules. Downloading just the source as a zip causes our builds to break since the zip does not include the submodules. We had a different user have the same experience #629 

I'm closing this issue since we already have #629 tracking a fix for the multiple users who have run into this issue.

closed 2019-08-19 by schultetwin1