microsoft/Azure-Kinect-Sensor-SDK #659: depthengine.dll for windows 7

KaiGuo-Vision, 2019-08-19:
Hello,

I am building the SDK source code under Windows 7, everything is going well except I could not configure the depthengine.dll in my Windows 7 system. 

I checked the relevant materials, looks like we could only get depthengine.dll by installing Azure Kinect SDK 1.0.*.msi, but this *.msi only supports Windows 10 system.  I also tried to copy depthengine.dll from friend's Windows 10 to my PC, but it does not work.

Would it be possible to share/release a depthengine.dll version for Windows 7 system?

I believe that lots of developer are still using Windows 7 system.

Best regards,

wes-b, 2019-08-19:
Sorry, we don't currently have any plans to support Win7. You can go to https://aka.ms/azurekinectfeedback and share your voice for this feature.

closed 2019-08-19 by wes-b

wes-b, 2019-08-19:
#630

yuemasy, 2019-10-11:
I tried k4aviewer tool on win10 version 1607 which is older than the System requirement and it doesn't work. Will the older win10 version be supported in future? Or what can i do to make it support that older win10 version? @wes-b @tesych 
Best regards,

wes-b, 2019-10-11:
Sorry, we don't have any plans to support older versions of Windows at this point in time.

You can file a request for support of specific OS versions over at our user feedback site.

wes-b, 2019-10-11:
Opps, sorry for the same answer twice @yuemasy. I forgot about the other conversation on this thread.

Win 10 1607 is older than our min version of Win 10 1803. If you can repro a failure on 1803 then please open a new bug and share the logs.