microsoft/Azure-Kinect-Sensor-SDK #660: No microphone recording - k4arecorder

somacoder, 2019-08-19:
Much to my disappointment, I just realized that the native k4arecorder application has absolutely no support for recording audio via the microphone arrays. What's more, on the external syncronization page for the k4arecorder here: https://docs.microsoft.com/en-us/azure/kinect-dk/record-external-synchronized-units it mentions to "Disable the microphone and IMU if they aren't needed to improve reliability." However, there was no microphone support to begin with unless I am missing something.

What is the reason for this and will this be added anytime soon? It's obviously a crushing blow to out of box functionality of this Kinect, especially given its 7 microphone array is a very useful hardware feature.

tesych, 2019-08-21:
@somacoder, I am sorry you are disappointed, but we do not have a plan at the moment to add support for recording audio since there are multiple tools available to do so. However, you can always add your request to [Customer Voice](https://feedback.azure.com/forums/920053) and if it gets  enough votes, we will definitely consider adding this functionality.

closed 2019-08-21 by tesych

somacoder, 2019-08-21:
@tesych Thanks for your response. The bottom line is there is no ability of your API to record video and audio simultaneously from a camera. Whether or not another program can separately access the microphones through DirectShow or WASAPI is irrelevant. If it's not natively synced with the video through the API recorder then it doesn't matter, especially with multiple cameras, where that desynchronization will only compound. What's worse, is that that relevant programming for synchronous microphone access is available across platforms on the viewer, but is not mirrored in the recorder, so what's the point? It's a bug.

UnaNancyOwen, 2019-09-27:
I think this is one of the features that should be added. 👍 

> I am sorry you are disappointed, but we do not have a plan at the moment to add support for recording audio since there are multiple tools available to do so.

@tesych What are the tools? Please tell us typical tools name. Thanks,

1216704928, 2019-11-21:
I had the same problem, how do we record audio?

xiaotaw, 2020-04-14:
I tried a third party library odas for audio recording: https://github.com/xiaotaw/odas.git.

Hope it helps.