microsoft/Azure-Kinect-Sensor-SDK #667: IMU example

rfilkov, 2019-08-22:
**Is your feature request related to a problem? Please describe.**

Currently Sensor SDK provides only raw IMU data. There is no API or example, how to convert the IMU data to device rotation and change in position.

**Describe the solution you'd like**

Please provide SDK API or at least example how to get the device rotation and change in position, according to the raw IMU data coming from device.

**Describe alternatives you've considered**

**Additional context**

rfilkov, 2019-08-28:
Just published the feature request here: https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38471407-imu-example-or-api I'm not sure though, if this kind of "feature" need to be explicitly requested.

prisonerjohn, 2019-09-01:
Agreed this would be very useful. Do we need to perform any calibration or correction or is this already handled by the SDK? An example would help answer these questions.