microsoft/Azure-Kinect-Sensor-SDK #668: Get blurry color image and all black depth image

opalmu, 2019-08-22:
**Describe the bug**
I tried to store color images and depth images captured from Kinect Azure using opencv for later application with Neural Networks. I just placed the camera pointing at the objects with the proper range. However my code just displayed blurry color image and all black depth image. 

**To Reproduce**
To reproduce the result I attached my code here.
```
// Copyright (c) Microsoft Corporation. All rights reserved.
// Licensed under the MIT License.

#include <iostream>
#include <stdio.h>
#include <stdlib.h>
#include <k4a/k4a.h>
#include <opencv2/core/mat.hpp>
#include <opencv2/core/core.hpp>
#include <opencv2/highgui/highgui.hpp>

using namespace cv;
using namespace std;

int main()
{
	// 1.1: Start by counting the number of connected devices
	uint32_t device_count = k4a_device_get_installed_count();
	if (device_count == 0)
	{
		printf("No K4A devices found\n");
		return 0;
	}
	else
	{
		printf("Found %d connected devices:\n", device_count);
	}

	// 1.2: Define the Exit block
	int returnCode = 1;

	// 1.3 Initialize the the device and capture attributes
	k4a_device_t device = NULL;
	k4a_capture_t capture = NULL;
	const int32_t TIMEOUT_IN_MS = 1000;

	// 1.4: Initialize the frame count
	int totalFrame = 5;
	int captureFrameCount = totalFrame;
	printf("Capturing %d frames\n", captureFrameCount);

	// 2: Set the configuration of device, you can also set it after open the device but before starting the camera
	k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
	config.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;  // <==== For Color image
	config.color_resolution = K4A_COLOR_RESOLUTION_2160P;
	config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;  // <==== For Depth image
	config.camera_fps = K4A_FRAMES_PER_SECOND_30;
	config.synchronized_images_only = true;

	// 3: Open the device
	if (K4A_RESULT_SUCCEEDED != k4a_device_open(K4A_DEVICE_DEFAULT, &device))
	{
		printf("Failed to open device\n");
		goto Exit;
	}

	// 4: Start the camera
	if (K4A_RESULT_SUCCEEDED != k4a_device_start_cameras(device, &config))
	{
		printf("Failed to start device\n");
		goto Exit;
	}

	// 5: Start to receive the captures and framess
	while (captureFrameCount-- > 0)
	{
		// Get a depth frame
		switch (k4a_device_get_capture(device, &capture, TIMEOUT_IN_MS))
		{
		case K4A_WAIT_RESULT_SUCCEEDED:
			break;
		case K4A_WAIT_RESULT_TIMEOUT:
			printf("Timed out waiting for a capture\n");
			continue;
			break;
		case K4A_WAIT_RESULT_FAILED:
			printf("Failed to read a capture\n");
			goto Exit;
		}

		printf("Capture %4d\n", captureFrameCount);

		// 5.1 Probe for a color image
		k4a_image_t image = k4a_capture_get_color_image(capture);
		if (image)
		{
			// 5.2 Get the sizes of color image
			int width = k4a_image_get_width_pixels(image);
			int height = k4a_image_get_height_pixels(image);
			int strides = k4a_image_get_stride_bytes(image);

			// 5.3 Store the image using opencv Mat
			uint8_t* color_image_data = k4a_image_get_buffer(image);
			const Mat color_image(width, height, CV_8UC4, (void*)color_image_data, Mat::AUTO_STEP);

			// 5.5 Display the images
			namedWindow("foobar", WINDOW_AUTOSIZE);
			imshow("foobar", color_image);
			waitKey(3000);

			// 5.6 release the image
			k4a_image_release(image);
		}
		else
		{
			printf(" | Color None                       ");
		}

		// 6.1 Probe for a depth16 image
		k4a_image_t image_depth = k4a_capture_get_depth_image(capture);
		if (image != NULL)
		{
			// 6.2 Get the sizes of depth image
			int width = k4a_image_get_width_pixels(image_depth);
			int height = k4a_image_get_height_pixels(image_depth);
			int strides = k4a_image_get_stride_bytes(image_depth);

			// 6.3 Store the image using opencv Mat
			uint8_t* depth_image_data = k4a_image_get_buffer(image_depth);
			const Mat depth_image(width, height, CV_16U, (void*)depth_image_data, Mat::AUTO_STEP);

			// 6.5 Display the images
			namedWindow("foobar", WINDOW_AUTOSIZE);
			imshow("foobar", depth_image);
			waitKey(3000);

			// 6.6 release the image
			k4a_image_release(image_depth);
		}
		else
		{
			printf(" | Depth16 None\n");
		}

		// release capture
		k4a_capture_release(capture);
	}

	returnCode = 0;
Exit:
	if (device != NULL)
	{
		k4a_device_stop_cameras(device);
		k4a_device_close(device);
	}

	return returnCode;
}
```

**Expected behavior**
Clear depth and color images show up.

**Logs**

**Screenshots**
Color image example here:
[image]

Depth image example here:
[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 10
 - SDK Version:  1.1.1
 - Firmware version:  N/A

**Additional context**
My very first guess is I need to set the correct configuration for the camera, or do I need to calibrate the camera in my code? Many thanks for your help!

mbleyer, 2019-08-22:
Please look at the constructor of CV::Mat:

Mat (int rows, int cols, int type)

Seems that you are passing in width instead of height and vice versa.

opalmu, 2019-08-22:
@mbleyer Thanks you are correct! I modified the code and it is fine now:-).

closed 2019-08-22 by opalmu