microsoft/Azure-Kinect-Sensor-SDK #671: (On Ubuntu 16.04) How to install k4a library after building it from source

msr-peng, 2019-08-23:
The problem still exists in the last SDK.
Not already resolved

**Describe the bug**
I tried to install the k4a library (on Ubuntu 16.04, to make it as a shared library for my other ROS project) after following the instruction here (https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) to build the Azure Kinect SDK. I can succesfully run *k4aviewer* and get the RGB-D stream after building Azure Kinect SDK. But then the installation failed after I run `ninja install` in `build` folder.

**To Reproduce**
```
git clone https://github.com/microsoft/Azure-Kinect-Sensor-SDK.git
cd Azure-Kinect-Sensor-SDK
mkdir build && cd build
cmake .. -GNinja
ninja
ninja install
```

**Expected behavior**
It should install the k4a library in `usr/local` directory.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  

**Additional context**
I think it's necessary to enable Azure Kinect to work in Ubuntu 16.04 & ROS environment. Since Azure Kinect targeted customers include robotics researchers. But the fact is that many of robots are tied to Ubuntu 16.04. Upgrading the robot hardware to 18.04 is difficult and would break existing projects. Although upgrading to 18.04 will eventually be necessary, it is helpful to have the Azure Kinect DK working on 16.04 in the meantime.

liaoyinyu, 2019-08-23:
you can try this project [https://github.com/microsoft/Azure_Kinect_ROS_Driver](url) , maybe it can work at Ubuntu 16.04.

RoseFlunder, 2019-08-23:
@liaoyinyu
The ROS package in the linked repository needs an installed azure kinect sensor sdk.
So its only tested with Ubuntu 18.04. with ROS Melodic.

msr-peng, 2019-08-23:
> you can try this project [https://github.com/microsoft/Azure_Kinect_ROS_Driver](url) , maybe it can work at Ubuntu 16.04.

Thank you. I know this link. The problem is that the ROS package now can only work on Ubuntu 18.04 (which install azure kinect sensor sdk by apt pacakge manager). But for Ubuntu 16.04, the only way I found to make azure kinect sensor sdk work is building it from source. And now I can't install kinect sdk (required by the ROS package) after building it.

msr-peng, 2019-08-24:
Oh, I forgot to add 'sudo' to do installation. Now I successfully installed the Azure Kinect SDK on Ubuntu 16.04.

Provide my workaround to make Azure Kinect work in Ubuntu 16.04 & Ros Kinetic environment:
     
On Ubuntu 18.04:
- Instill k4a-tools and libk4a1.1-dev from apt package manager, then get `libdepthengine.so.1.0` and `libstdc++.so.6` from your installation path (for me it is /usr/lib/x86_64-linux-gnu/).
     
On Ubuntu 16.04:
- Build Azure Kinect SDK from source (be careful that it won't work for the lastest SDK, so you need git checkout Azure Kinect SDK repository to commit ID *9afcd78*). Then install SDK.
- Copy the two SDK .so files (`libdepthengine.so.1.0` and `libstdc++.so.6`) from Ubuntu 18.04 to your lib subdirectory of SDK installation path (for me it is /usr/local/lib/).
- Enable to use Azure Kinect SDK with the camera without being 'root' by setting up udev rules: cp Azure-Kinect-Sensor-SDK/scripts/99-k4a.rules /etc/udev/rules.d/
- Git clone Azure Kinect ROS package (https://github.com/microsoft/Azure_Kinect_ROS_Driver.git) to your ROS Kinetic workspace (`catkin_ws`)

closed 2019-08-24 by msr-peng

caselitz, 2019-12-06:
I had to additionally copy `libm.so.6` (from /lib/x86_64-linux-gnu/) from Ubuntu 18.04 to the lib subdirectory of SDK installation path (for me it is /home/user/Azure-Kinect-Sensor-SDK/build/bin).

This is for SDK Version 1.4.0 and `libdepthengine.so.2.0`