microsoft/Azure-Kinect-Sensor-SDK #672: "MJPEG decoder initialization failed" on Ubuntu 18.04

skalldri, 2019-08-23:
**Describe the bug**

Users of the Azure Kinect ROS Driver are seeing issues where the MJPEG decoder fails to initialize. Clearing this error requires restarting the PC. The only error provided by the SDK is:

```
[2019-08-23 10:54:16.941] [error] [t=13655] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/color/uvc_camerareader.cpp (1267): DecodeMJPEGtoBGRA32(). MJPEG decoder initialization failed

[2019-08-23 10:54:16.941] [error] [t=13656] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (271): capturesync_get_capture(device->capturesync, capture_handle, timeout_in_ms) returned failure in k4a_device_get_capture()
```

[Based on the code that generates the error](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/eaedb7aaa295367062645c501f5df5f383d53ce7/src/color/uvc_camerareader.cpp#L1280-L1285), this appears to be a fatal error in libjpeg. 

**To Reproduce**

We aren't 100% clear on the repro steps for this issue. The best we can tell, we just need to restart an application that uses the color camera in BGRA mode lots of times, and eventually this issue will reproduce.

**Expected behavior**
MJPEG decoder initialization should never fail.

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04
 - SDK Version:  1.1.1
 - Firmware version: 
```
 == Azure Kinect DK Firmware Tool == 
Device Serial Number: 000134685112
Current Firmware Versions:
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft
```

**Additional context**

skalldri, 2019-08-23:
It looks like libjpeg can provide more details on the error if we call `tjGetErrorStr()`. At a minimum, the SDK should be calling that function and providing a more verbose libjpeg error to the user.

skalldri, 2019-08-23:
I added a line to print the contents of tjGetErrorStr() and I'm getting this as the output:
```
[2019-08-23 11:31:25.386] [error] [t=7529] ../src/color/uvc_camerareader.cpp (1268): DecodeMJPEGtoBGRA32(). tjGetErrorStr(): Wrong JPEG library version: library is 80, caller expects 62
```

Any idea what this means?

closed 2019-09-13 by schultetwin1

crypto-g245, 2022-02-09:
Hello, 

I was trying to connect the cameras on Intel NUC with addiitional [startech PCI card ](https://www.amazon.de/StarTech-com-Express-SuperSpeed-Dedicated-Channels/dp/B00HJZEA2S). But I am running into the same error. 

OS with Version: Ubuntu 18.04
SDK Version: 1.4.1
color image format: BGRA 
Firmware version:
[ info     ] : ******************** Device Info ********************
[ info     ] : K4A SDK version:     1.4.1
[ info     ] : Serial Number:       000398104412
[ info     ] : RGB Sensor Version:  1.6.110
[ info     ] : Depth Sensor Version:1.6.79
[ info     ] : Mic Array Version:   1.6.14
[ info     ] : Sensor Config:       6109.7
[ info     ] : Build type:          Release
[ info     ] : Signature type:      MSFT
[ info     ] : ****************************************************

Log from the k4aviewer is attached here . [k4a_log.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/8031378/k4a_log.txt)

With little debugging I have found that it could be due to tjDecompress2  which comes from this [libjpeg/turbojpeg](https://jpeg-turbo.dpldocs.info/libjpeg.turbojpeg.tjDecompress2.html) 

[image]

Is there a specific version for this library which could be a fix?

[k4a_log.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/8031378/k4a_log.txt)

PR #763 (2019-09-13): Properly Sets default visibility of functions
## Fixes #672 

### Description of the changes:
- Properly sets visibility flags for GCC / Clang

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [ ] I ran the [unit tests](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/testing.md)
- [ ] I ran the [functional tests](https://github.com/Microsoft/Azure [... cut]