microsoft/Azure-Kinect-Sensor-SDK #678: How to get the parameters of depth camera?

Miss-Dora, 2019-08-26:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

Miss-Dora, 2019-08-27:
> @Miss-Dora, can you please provide more information about what exactly you are looking for?
I want to use Azure Kinect to get undistorted depth image, and the camera parameters corresponding to the undistorted depth image. I need convert depth image to pointcloud to do some processing.

Miss-Dora, 2019-08-27:
> You may find the documentation on [`k4a_calibration_intrinsic_parameters_t`](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/unionk4a__calibration__intrinsic__parameters__t.html) helpful (related: [`k4a_calibration_t`](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/structk4a__calibration__t.html), [`k4a_calibration_intrinsics_t`](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/structk4a__calibration__intrinsics__t.html) ), if that's what you're looking for. If it's not, please provide more info and we'll be happy to help!
I want to use Azure Kinect to get undistorted depth image, and the camera parameters corresponding to the undistorted depth image.Now I have connected Azure Kinect to computer, and used "k4aviewer.exe" get a deep data named "xx.mkv",I need the camera parameters corresponding to the depth image,they are fx,fy,cx,cy. How can I get them?

rabbitdaxi, 2019-08-29:
@Miss-Dora have we answered your questions? If yes, please close this issue :) Thank you for the feedback.

closed 2019-08-31 by rabbitdaxi