microsoft/Azure-Kinect-Sensor-SDK #679: Throw exception when calling k4abt_tracker_create

DogeDan, 2019-08-26:
Hello, I have an issue with the body tracking tutorial code. When I debug this project, I always get an exception at the certain function: `k4abt_tracker_create(&sensor_calibration, &tracker)`, so that I can't create a body tracker to do the following step. 

> Exception thrown at 0x00007FFD9F58B1E8 (k4abt.dll) in BodyTrackingTutorial.exe: 0xC0000005: Access violation reading location 0x0000000000000083.

How can I fix it?
Thank you

SDK Version: 1.1.0
Body Tracking SDK Version: 0.9.1

DogeDan, 2019-08-27:
> Hi @DogeDan, thanks for reporting this issue. May I ask what NVIDIA driver version your PC is using?
Hi @yijiew , below is the NVIDIA driver information:
[image]

An interesting thing is that the program works when I run directly the execution file in the debug folder, but it always throws the exception when I use Debugger on VS2015.

DogeDan, 2019-08-28:
Hi @yijiew , thanks for your reply. It works after I copied the dnn_model.onnx file to my code folder! Thanks again.

rfilkov, 2019-08-28:
Hi @yijiew , first off congrats on the decision to separate the body tracking SDK from the sensor SDK. I managed to make it even work with RealSense sensors. To my surprise though, the body tracking works OK, while the body index image stays empty. What may be the reason for this?

rfilkov, 2019-08-28:
> Thanks for letting us know that it works on different sensor! Supporting other sensors is not in our scope but I am personally interested in seeing these results. However, since it is not related to this issue, could you create a new issue with "BodyTracking" tag and assign to me? We can continue our discussion there.

@yijiew OK I'll do it, although I'm not sure how to assign you to the new bug report. I can send you the package I'm working on (if you'd like to take a look), but please tell me an e-mail address to send it via WeTransfer, or some other way to privately share it with you.

rfilkov, 2019-08-28:
@yijiew here it is: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/693

closed 2019-08-29 by DogeDan