microsoft/Azure-Kinect-Sensor-SDK #687: The joint orientation of body tracking is confusing.

venscn, 2019-08-28:
**Describe**
What is the initial value of the joint orientation? It is confusing, and hard to adjust to the right look.

**Feature**
Maybe can use the same description as kinect v2, that will be very clear, and it is not necessary to adjust the posture according to the direction.

rfilkov, 2019-08-28:
https://docs.microsoft.com/en-us/azure/kinect-dk/body-joints

wqs111000, 2019-08-28:
It is similar to the issue #654.The replies there may help you.

venscn, 2019-08-28:
@wqs111000
I mean the left arm, right arm, left leg, right leg is not uniform with respect to the torso. I want to controller a human model, but the coordinate system makes me a headache.

qm13, 2019-12-02:
We predominantly use synthetic data to train the body tracking model using rigged avatar models. The axis orientation is widely used with commercial avatars to support 3 main use cases: posing, animating (which is like posing just with keyframes obviously) and applying motion capture data.

Posing and animation require the developer to work with the joints/bones. A large part of this work involves just modifying rotations. The axis orientation makes it easy to make mirrored movements e.g. raise both arms by 20 degrees.

Most game engines and rendering software e.g. Maya also use axis orientation. We also decided to use axis orientation with Azure Kinect Body Tracking SDK. You can implement a simple translation to covert to the legacy Kinect for Windows orientation. If you are interested in a future version of the SDK support the legacy orientation please open feedback at https://feedback.azure.com/forums/920053-azure-kinect-dk.

closed 2019-12-02 by qm13