microsoft/Azure-Kinect-Sensor-SDK #688: Depth image correction in NFOV unbinned mode

ivaandrey, 2019-08-28:
Hello, 
How can I correct the distortion of depth image in NFOV unbinned mode?
Does 3D cloud data  contain the undistorted depth data. How can I extract the undistorted depth image from 3D cloud data?
Thanks

ivaandrey, 2019-08-28:
Ok, thank you for the quick answer.
What function should I run to get 3d data from 2d data? Does the transformation from 2d to 3d perform the same undistortion logic you sent me in example?Or it other undistorion method?
Thanks

ivaandrey, 2019-08-28:
I see, is there any way to extract the undistorted depth image from 3d data?

rabbitdaxi, 2019-08-29:
@ivaandrey have we answered your questions? If yes, please close this issue :) Thank you for the feedback.

ivaandrey, 2019-08-30:
Yes, thank you a lot

closed 2019-08-30 by ivaandrey