microsoft/Azure-Kinect-Sensor-SDK #689: C# wrapper of Body Tracking SDK

thorikawa, 2019-08-28:
**Is your feature request related to a problem? Please describe.**

While the latest version of Sensor SDK has C# wrappers, Body Tracking SDK does not. It make hard to use Body Tracking SDK from other frameworks/game engines like Unity.

**Describe the solution you'd like**

I think it is better to have C# wrappers of Body Tracking SDK in this repo. I myself wrote them in https://github.com/curiosity-inc/azure-kinect-dk-unity/ and happy to merge them if this request is approved.

**Describe alternatives you've considered**

If having Body Tracking SDK wrappers in this repo is not desirable for some reasons, I suggest that make `Microsoft.Azure.Kinect.Sensor.NativeMethods` class available to external (currently it is internal class), so that we can easily extend the existing SDK to have Body Tracking SDK.

**Additional context**

Adding C++ wrappers for Body Tracking SDK is requested [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/561).

Necromantic, 2019-08-30:
I've added similar rudimentary body tracking functionality to the C# wrapper that works with version 1.2 up to todays commits.
It's a quick and dirty extension to the current wrapper, use at your own risk. ;)
https://github.com/Necromantic/Azure-Kinect-Sensor-SDK

wes-b, 2019-09-09:
Lets please track this request over at https://aka.ms/azurekinectfeedback.

closed 2019-09-09 by qm13

Necromantic, 2019-10-17:
Sorry, I wasn't aware it would disable the Issues on a forked Repo by default. I've enabled them now.
It should have been Kinect SDK 1.2.0 and Azure Body Tracking 0.9.2 back then.
I will update it to 1.3.0 and 0.9.4 as soon as I'll find some time for it.
For further discussion we may move over to my fork.