microsoft/Azure-Kinect-Sensor-SDK #693: Body tracking with RealSense

rfilkov, 2019-08-28:
**Describe the bug**

@yijiew , I managed to make the K4A Body Tracking SDK work with RealSense sensors. To my surprise though, while the body tracking works OK, the body index image stays empty. What may be the reason for this?

**To Reproduce**

I can provide you the repro package.

**Expected behavior**

The body index map to reflect the tracked bodies.

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

yijiew, 2019-08-28:
@rfilkov No worry. Take your time. I will give it a try tomorrow then :)

rfilkov, 2019-08-28:
@yijiew OK, thank you! I'll send it tomorrow, if you don't mind. Together with some clips for testing, as requested. It's almost midnight here :)

rfilkov, 2019-08-29:
@yijiew I sent the package to the provided e-mail address, along with some instructions. Please tell me, if you need more info.

qm13, 2019-08-30:
Microsoft Azure Kinect Body Tracking does not support the RealSense range of depth cameras. Please create a feedback at https://feedback.azure.com/forums/920053-azure-kinect-dk if you are interested in RealSense support.

closed 2019-08-30 by qm13