microsoft/Azure-Kinect-Sensor-SDK #698: Most efficient way to get undistorted & registered RGB-D images

fishcu, 2019-08-29:
What is the most efficient way, from a pipeline view, to generate registered (overlapping) color and depth images, which are also undistorted (that is, conforming to a pinhole projection)? I would like to map the depth to color, not the other way round.

Secondly, what would be a good way to get the corresponding pinhole parameters (if not calculated as part of the rectification)?

I went through the API and examples (specifically, the "undistort" example). One way which seems possible:

Initialization:
- Get calibration
- Create transformation from calibration
- Use `create_pinhole_from_xy_range` from `undistort/main.cpp`, but use color camera instead of depth
- Create a LUT as in `create_undistortion_lut` from `undistort/main.cpp`, again using color camera

Main loop:
- Capture frames
- Map depth to color using `transformation.depth_image_to_color_camera`
- Remap both color and depth images using the LUT we created above.

The advantage of that pipeline is that all the code is already there (from examples and the SDK itself). The disadvantage is that some things are not very efficient, for example the depth is essentially un-projected, then reprojected, then remapped again. It would be better to project it directly using the correct pinhole calibration.

Any help is appreciated.

fishcu, 2019-08-29:
Another comment I have is that I don't understand the need for interpolation of the depth map. It seems wrong to interpolate depth, as it will introduce artifacts around depth discontinuities. There is a code example which at least excludes interpolating between valid and invalid data, but I guess a threshold parameter would make more sense: Only interpolate if data is sufficiently close.

rabbitdaxi, 2019-08-29:
@fishcu , thank you for the feedback and there are some comments to your two points:
1) Yes, your pipeline sounds reasonable to get the registered RGB-D images and also undistorted both of them in the color camera domain. I think that should be the right thing to do with the SDK API. Regarding performance, the depth to color transform is on GPU, therefore, that leaves the undistort part of remapping both images in color resolution on CPU in runtime, hope it can reach to your performance requirement. 

2) the undistort example initially was only support nearest neighbor, which we agree that will definitely won't introduce new depth value. some people feel NN bring the quantization artifacts, and they want a more smooth depth map, therefore they start using opencv undistort functions with LINEAR interpolation. Our undistort example mainly want to showcase to user that the opencv LINEAR will introduce the noise on the depth discontinuity when there are valid and invalid depth pixels (we check that using the bilinear interpolation in the undistort example), but totally agree that we should also add a threshold to avoid interpolate if all neighbors are valid and there is depth discontinuity in there. And we will add that threshold before close this issue. Good catch!

closed 2019-08-31 by rabbitdaxi

T-Captum, 2021-05-12:
I am doubtful the pipeline as proposed by fishcu will actually result in undistorted images. Here is why: 
When using `transformation.depth_image_to_color_camera` the depth image is transformed into the color camera space. However the depth image has a different distortion than the color camera. And afterwards we only undistort with respect to the color camera. For this to result in an undistorted depth image the `transformation.depth_image_to_color_camera` function would have to undistort the depth image with respect to the depth camera during the transformation and apply the color camera transformation afterwards. Does this actually happen? 

Thanks!

LuyaooChen, 2021-07-17:
Hi, I think the `remap` function in `example/undistort` can't be applied to color images, right?  Because it use `uint16_t` to access iamge data. Any adevice to undistort color image?