microsoft/Azure-Kinect-Sensor-SDK #699: CMakeLists.txt incorrectly detects x86 vs amd64

schultetwin1, 2019-08-29:
**Describe the bug**
In our CMakeLists.txt files we use the following logic to detect if we are building for x86 versus amd64.

```cmake
if ("${CMAKE_SIZEOF_VOID_P}" STREQUAL "8")
    set(TARGET_ARCH "amd64")
elseif ("${CMAKE_SIZEOF_VOID_P}" STREQUAL "4")
    set(TARGET_ARCH "x86")
endif()
```

This works great while we are building for x86 and amd64 only. However, if we decide to support arm64 this will break.

We have not yet decided to support arm or not, but regardless, this should be fixed just in case we decide to go forward with arm support.

wes-b, 2020-04-14:
@schultetwin1 This is fixed now with the arm64 chance IIRC?

schultetwin1, 2020-04-14:
You are correct! This was fixed by #1108

closed 2020-04-14 by schultetwin1