microsoft/Azure-Kinect-Sensor-SDK #707: How to improve the depth reading in the corners of a room?

lordjesus, 2019-09-02:
I am using the Azure Kinect with the wide field-of-view depth camera to detect objects being put on the floor, by first getting a reading of the depth values with no objects, and then comparing the depth of the new captures to get the difference. The problem I am having is that the depth reading in the corners of the room is very bad. I have been reading this document https://docs.microsoft.com/en-us/azure/kinect-dk/depth-camera#invalidation and know that there can be difficulties with corners of rooms. 

What I want to know is if there is anything I can do to improve the depth readings, either through configurations of the camera or other tweaks. I have to use the WFOV depth camera, the narrow one is too narrow for my use case.

Below are some screenshots of the scene that caused me trouble. In this case the pixels are invalidated almost a meter into the room, especially in the corners.

[image]
[image]
[image]

rabbitdaxi, 2019-09-04:
@lordjesus , thank you for the feedback. Quick questions and experiment to try:
- are you using WFOV_2X2BINNED or WFOV_UNBINNED, you can try both to see whether they make are difference
- I see the bright edge in the IR image at the bottom, is that a table? Can you try to lift the camera higher or put the camera on a tripod see whether it improves your reading at the corners in the room?
- If you move the camera closer to the scene, will that improve the corner? 

The invalidation at the corner could be because of low signal or multipath detection, during either case, the depth value accuracy could be out of spec, therefore, the invalidation is trying to ensure the depth you got from the camera is accurate. 

Another idea you can try could be 2 or more cameras with NFOV

rabbitdaxi, 2019-09-13:
@lordjesus hope the answers above help you, let us know if you have more questions.

closed 2019-09-13 by rabbitdaxi