microsoft/Azure-Kinect-Sensor-SDK #708: Body tracker configuration sensor orientation no changes

nicolascarre, 2019-09-02:
I just updated the body tracker, now there is a body tracker config (k4abt_tracker_configuration_t).
I use it like the sensor config, before create the body tracker, I set up the body tracker config, in my case, I set "K4ABT_SENSOR_ORIENTATION_FLIP180" for the sensor orientation.
I put the camera upside down, but there is absolutly no changes.

Is that normal ?
What do I did wrong ?
What is the expected behavior ?

wes-b, 2019-09-09:
Can you please tell us what your expected results were and what you actually got?

nicolascarre, 2019-09-10:
What I got is nothing different with any config.
I thought that if I put the camera upside down set the config with "K4ABT_SENSOR_ORIENTATION_FLIP180", the body tracker would detect the skeleton perfectly.
Say differently, I guessed you reversed the images (color, depth, IR) before detecting a skeleton, or something like that.

yijiew, 2019-09-16:
@nicolascarre Thanks for reporting this issue. The problem will be solved in the next release.

yijiew, 2019-09-20:
The new 0.9.3 release that solves this problem is published. You can download the latest release here: 0.9.3 | [msi](https://www.microsoft.com/en-us/download/details.aspx?id=100307) [nuget](https://www.nuget.org/packages/Microsoft.Azure.Kinect.BodyTracking/0.9.3)

closed 2019-09-20 by yijiew