microsoft/Azure-Kinect-Sensor-SDK #712: Image stride in k4aviewer "too large"?

fishcu, 2019-09-03:
Minor thing... Just came across this line in `tools/k4aviewer/k4apointcloudvisualizer.cpp`, lines 140ff:

```
m_pointCloudColorization = k4a::image::create(K4A_IMAGE_FORMAT_COLOR_BGRA32,
                                                      m_calibrationData.depth_camera_calibration.resolution_width,
                                                      m_calibrationData.depth_camera_calibration.resolution_height,
                                                      m_calibrationData.depth_camera_calibration.resolution_width * 3 *
                                                          static_cast<int>(sizeof(int16_t)));
```

It seems like it should be something like `sizeof(BgraPixel)` instead (dropping the 3 times multiplier)? Possibly a bug without any resulting issues?

wes-b, 2019-09-04:
@rabbitdaxi should this be K4A_IMAGE_FORMAT_CUSTOM instead of BGRA?

rabbitdaxi, 2019-09-04:
@fishcu good catch, by looking at the code, `sizeof(BgraPixel)` definitely makes more sense

PR #717 (2019-09-05): Fix the point cloud colorization image stride

## Fixes #712 

### Description of the changes:
- Fix the point cloud colorization image stride

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [x] I ran the [unit tests](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/testing.md)
- [x] I ran the [functional tests](https://github.com/Microsoft/Azure [... cut]
closed 2019-09-05 by wes-b