microsoft/Azure-Kinect-Sensor-SDK #714: Orientation of Kinect seems to affect depth results.

WheezardX, 2019-09-03:
**Adding depth readings from multiple Kinects into the same Unity scene, the physical orientation of the Kinects appear to affect results.  Even after transformation is applied.**

**To Reproduce**
1. Connect multiple Kinects to the same machine.  
2. Create a Unity scene with game objects that represent the physical sensors.
3. Attach a script that is capable of rendering the point cloud.

Correct Case
1.  Set Kinects next to each other, same orientation.
2.  Update Unity objects with the correct transformations
3.  Render results in Unity. (Or preferred renderer)
4.  Notice point cloud is similar, with only horizontal offsets that represents the physical distance between the sensors.

Incorrect Case
1.  Set Kinects next to each other, with one up-side down.
2. Update Unity objects with the correct transformations
3.  Render results in Unity.  (Or preferred renderer)
4. Notice point clouds are now offset on the vertical axis and also contain some tilt.

**Expected behavior**
Depth readings should not be affected by orientation of Kinect sensor when using k4a_transformation_depth_image_to_point_cloud().

**Screenshots**
Kinects next to each other with same orientation.  Notice the horizontal offset is expected. 
[image]

Kinects in essentially the same position but one flipped over.  (Unity object also flipped to account for change in vertical axis)

[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 1809
 - SDK Version:  1.2
 - Firmware version:  

**Additional context**
I am not using the C# library.  I am using a DLL wrapper around one of the C++ examples I extended.  There is a function for initialization and a function for retrieving the point cloud.  It uses k4a_device_get_capture() to get the capture then gets the color and depth images from that.  Finally it uses k4a_transformation_color_image_to_depth_camera() and k4a_transformation_depth_image_to_point_cloud(), then steps through those buffers to write them to an output buffer used in the C# code by Unity.

rabbitdaxi, 2019-09-04:
@WheezardX , thank you for opening up the issue. 

Based on what I read from your email, you took two kinects and got the point cloud from each of them. 

Can you clarify:
Whether you just render them naively without bring the two point cloud in the same coordinate system?  (e.g. a common approach for multiple camera point-clouds rendering should be: you need to calibrate the extrinsics between the two Kinect depth cameras, you can refer to an [example](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/green_screen) which provides a reference of how to calibrate the extrinsics of two camera. You can do similar thing. Then once you have the extrinsics between two depth cameras, you can transform the point cloud from one depth camera to the other and render them in the same coordinate system, then they should be well aligned assuming your calibration is accurate. 

If you are not doing such a transformation and just render the two point cloud from there own coordinate system in the unity app, that will explain the horizontal offset when they are side by side (simply because the two point cloud has difference coordinate system origin). 
Then it also can explain the not "aligned" or "having an angle shift" when you flip the other camera given the two point clouds are essentially from two different coordinate systems.

If you are not render the two point cloud from the same coordinate system, you should do so.

WheezardX, 2019-09-04:
I am accounting for the extrinsics via the transformation system in Unity.  For example, when I flip a camera over, I flip it over in the Unity scene as well.  That is why the point clouds are only off by a few degrees and not 180 degrees.   The horizontal offset I wasn't worried about because that was expected (I intentionally wasn't accounting for it).

In my primary case, I am using 4 Kinects mounting on a physical frame at opposite corners facing inward.  This frame was built in CAD and we 3D printed mounts for the Kinects.  So we have a high degree of precision with respect to the camera's physical position and orientation inside of Unity.

However, our approach assumes the view direction of the camera is straight out from the Kinect body.  In reality, it looks like the camera lens may be shifted slightly downward like a projector often is.  

Here is an example shot from the Unity scene with the transformation gizmo aligned to the transformation (extrinsics) of the camera per our assumption.

[image]

It would appear the camera is more like this (somewhat exaggerated for demonstration purposes)
[image]

```   
    k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
    config.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;
    config.color_resolution = K4A_COLOR_RESOLUTION_720P;
    config.depth_mode = K4A_DEPTH_MODE_WFOV_2X2BINNED;
    config.camera_fps = K4A_FRAMES_PER_SECOND_30;
    config.synchronized_images_only = true; 
```
That is my configuration settings.  So I believe the answer to your question about depth mode is WFOV Binned.

The green screen example is interesting and we previously were using OpenCV with aruco markers to try and position the cameras in a fixed world space.  The problem was we were not getting very good resolution with that, so we went with the fixed frame approach.  Also, with 4 cameras in an orbit, it is impossible for all cameras to see the same marker, so multiple markers have to be used.

WheezardX, 2019-09-04:
I just noticed your edit with the link about the coordinate system.  This likely explains what I'm seeing.

"The depth camera is tilted 6 degrees downwards of the color camera, as shown below."

rabbitdaxi, 2019-09-04:
@WheezardX , yes, line I added is the explanation of what you saw when you flipped the camera, however, I do not think this is correct for you to align the point cloud from two depth camera, "I am accounting for the extrinsics via the transformation system in Unity. " 

If you render both point cloud in the unity, you need to choose a common coordinate system, it can be one of the two depth camera's coordinate system, let us say camera A is the coordinate system you should use. Then you need to calibrate the extrinsics between these two depth camera (when you have the relative location rigidly fixed, even they are flipped), then if you have such extrinsics, you can transform the point cloud from B to A with the extrinsics (rotation and translation), then you should see both point cloud aligned.

WheezardX, 2019-09-09:
Once we accounted for the downward tilt, I was able to co-locate 3/4 of the Kinects in a common world space.  The 4th seems to disagree a bit on depth but that may be a poor mount.  

Thanks for the help!

closed 2019-09-09 by WheezardX

yuemasy, 2020-03-11:
> I am accounting for the extrinsics via the transformation system in Unity.  For example, when I flip a camera over, I flip it over in the Unity scene as well.  That is why the point clouds are only [...]

Hi , when I measure depth value, I need the depth ground truth. I measure the distance from camera front cover to the object, and I need to know the distance from the  front cover to color sensor origin. Seems you use this value in your above picture. How do you know this value?  Thank a lot.

WheezardX, 2020-03-11:
IIRC the CAD was built from this.
https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification