microsoft/Azure-Kinect-Sensor-SDK #716: Build fails with K4A_VALIDATE_CLANG_FORMAT = ON

fishcu, 2019-09-04:
Latest develop branch commit.
Used regular CMake build generator (instead of Ninja).
Ubuntu 16.04.
GCC 8.

Build fails with the following error whenever `K4A_VALIDATE_CLANG_FORMAT` is turned on:
```
$ make
[  0%] Validating format of /home/user/tmp/Azure-Kinect-Sensor-SDK/src/csharp/Tests/StubGenerator/Stub.cpp
Traceback (most recent call last):
  File "/home/user/tmp/Azure-Kinect-Sensor-SDK/cmake/ValidateFormat.py", line 108, in <module>
    main(sys.argv[1:])
  File "/home/user/tmp/Azure-Kinect-Sensor-SDK/cmake/ValidateFormat.py", line 102, in main
    with open (args.output, 'w') as output:
FileNotFoundError: [Errno 2] No such file or directory: '/home/user/tmp/Azure-Kinect-Sensor-SDK/build/src/csharp/Tests/StubGenerator/Stub.cpp.clang-validate-result'
CMakeFiles/validateclangformat.dir/build.make:380: recipe for target 'src/csharp/Tests/StubGenerator/Stub.cpp.clang-validate-result' failed
make[2]: *** [src/csharp/Tests/StubGenerator/Stub.cpp.clang-validate-result] Error 1
CMakeFiles/Makefile2:117: recipe for target 'CMakeFiles/validateclangformat.dir/all' failed
make[1]: *** [CMakeFiles/validateclangformat.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
```

Build works fine when I turn off `K4A_VALIDATE_CLANG_FORMAT`.

fishcu, 2019-09-04:
One thing I should add: `clang-format` has version 6.0.0.

wes-b, 2019-09-10:
Unfotunately its working fine for me and your version is correct. If you just enabled this, you might need to start with a clean build folder.

fishcu, 2019-09-11:
I'm running into the same problem as above from a clean build folder. Is it working fine for you even when not using Ninja? That's one point where I deviate from the official instructions.

wes-b, 2019-09-11:
I don't know, we are only using Ninja internally. You are going to have dig into the issue unfortunately. It seems like the python script is having trouble opening the files for write.

jasjuang, 2019-12-03:
I just want to say that I run into the exact same problem. Temporally bypassing it with K4A_VALIDATE_CLANG_FORMAT=OFF for now.

limitz, 2020-04-27:
The problem is that the CMakeLists.txt in the root only creates the csharp folder if the compiler is msvc, while the generated cmakefile at ./build/CMakeFiles/clangformat.dir/build.make expects it to be there.