microsoft/Azure-Kinect-Sensor-SDK #721: segment fault while using the Qt creator to develop

alfredyu0911, 2019-09-05:
**Describe the bug**
segment fault when launch the application by IDE(Qt creator), but work fine if simply launch it by the executable file
        * [image]

**To Reproduce**
1. download and install [Qt creator 5.13.0](https://download.qt.io/archive/qt/5.13/5.13.0/)
    * installation detail
        1. [image]
2. download and unzip [the test project](https://drive.google.com/file/d/18SrzaAuR6h4YXcjVYKtKhcLRC5P6Rww-/view?usp=sharing) (it's about 480MB, because the whole SDKs is in it)
3. open the project by double click the file `bodytrackingTest.pro`
    * detail when first open this project
        1. [image]
        2. [image]
4. do as the figure [image]
5. do as the figure [image]
6. use the debug tool [image]
7. it should be like this after execute line #65 [image]

**Expected behavior**
run without crush

**Logs**
none

**Screenshots**
check Reproduce steps

**Desktop (please complete the following information):**
 - OS with Version: Windows 1809
 - SDK Version:  0.9.1
 - Firmware version:  1.1.0

**Additional context**
none

RoseFlunder, 2019-09-05:
Just a quick guess would be that the IDE starts the application in another folder (not the Debug folder where you placed the dll files) so that the IDE can't find the model and/or dll files.
Maybe you can check what your current working directory is when starting the application via IDE.
C++ 17 has a simple way to do it I think:
https://en.cppreference.com/w/cpp/filesystem/current_path

alfredyu0911, 2019-09-05:
> Just a quick guess would be that the IDE starts the application in another folder (not the Debug folder where you placed the dll files) so that the application can't find the model and/or dll files.
> Maybe you can check what your current working directory is when starting the application via IDE.
> C++ 17 has a simple way to do it I think:
> https://en.cppreference.com/w/cpp/filesystem/current_path

nope, not likely, because
1. if i just use the sensor SDK, then every thing work fine
2. mark the code which execute `k4abt_tracker_create`, and then it can launch by IDE without crush

yijiew, 2019-09-17:
Hi @alfredyu0911 , the guess from @RoseFlunder might be right. Because: 
1. The sensor SDK is totally separate from the body tracking SDK. So running sensor SDK correctly doesn't mean anything to the body tracking SDK. 
2. All the other dll dependencies are delay loaded in Windows. So if you comment out the body tracking SDK code, it won't hit any dependency issue. 

Start from the 0.9.2 version, the SDK will dynamically check the dependency existence. If they are not existed at the expected folder, it will fail to create the tracker and log the error instead of crashing. Could you give it a try to see whether it is the case?

alfredyu0911, 2019-09-18:
> Hi @alfredyu0911 , the guess from @RoseFlunder might be right. Because:
>
> 1. The sensor SDK is totally separate from the body tracking SDK. So running sensor SDK correctly doesn't mean anything to [...]

Hi @yijiew, thank you and @RoseFlunder :
I tried to change the working directory, and it did work, sorry that i didn't try it in the first place.
Thank you for your support.
FYI, the file `dnn_model.onnx` isn't checked dynamically (version  0.9.2), should i close this issue anyway?

yijiew, 2019-09-18:
@alfredyu0911 I am glad it solves your problem :D When you say "the file dnn_model.onnx isn't checked dynamically", do you mean if you don't have the dnn_model.onnx file at the correct location, it will still crash instead of just return error when you create the tracker?

alfredyu0911, 2019-09-19:
> @alfredyu0911 I am glad it solves your problem :D When you say "the file dnn_model.onnx isn't checked dynamically", do you mean if you don't have the dnn_model.onnx file at the correct location, it will still crash instead of just return error when you create the tracker?

@yijiew it return error when create the tracker, 
sorry, i read again and just realize it's not a issue, sorry to bother.
thank you.

closed 2019-09-19 by alfredyu0911

yijiew, 2019-09-19:
Thanks @alfredyu0911 !