microsoft/Azure-Kinect-Sensor-SDK #722: Azure Kinect Viewer GUI Blank

FreakTheMighty, 2019-09-05:
**Describe the bug**
After installing the latest sdk via the MSI the viewer is entirely blank.

**To Reproduce**
1. On a mac book pro with bootcamp install the SDK 1.2.0
1. Launch the kinect viewer

**Logs**
I'm not able to get the logs from the pre-built viewer because the UI isn't functional.

**Expected behavior**
The azure kinect viewer should display its typical UI
**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 Version 1803
 - SDK Version:  1.2.0
 - Firmware version:  
```
 == Azure Kinect DK Firmware Tool ==
Device Serial Number: 000222192512
Current Firmware Versions:
  RGB camera firmware:      1.6.98
  Depth camera firmware:    1.6.70
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft
```.

FreakTheMighty, 2019-09-05:
Also, I found that `.\k4arecorder.exe` appears to work. I haven't checked the contents, but it generates a mkv of a few hundred MB without any errors. This makes me think the issue lies with rendering the UI.

tesych, 2019-09-05:
@FreakTheMighty, Thank you for sharing with us, but at this point we do not support Macs. Please follow to [Customer Voice](https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38061958-please-allow-for-macos-support-should-be-minima) to vote for the feature.

FreakTheMighty, 2019-09-05:
Hi @tesych, to be clear I am running Windows 10, the feature request you linked is for macOS support. I mentioned the the hardware is a macbook running bootcamp incase that is relevant. Still, I am running Windows 10 Version 1803.

My laptop meets the hardware requirements mentioned here: https://docs.microsoft.com/en-us/azure/Kinect-dk/system-requirements

I humbly request we re-open this issue :D

wes-b, 2019-09-09:
Hi @FrankTheMighty, I suspect you have a bug with one of the UI frameworks and its compatibility with your graphics driver. 

Can you create a batch file launch K4aViewer.exe with ENV variable set? This will have th SDK collect its own log and hopefully give us a clue about what is happening.

```
set K4A_ENABLE_LOG_TO_A_FILE=k4a.log
set K4A_ENABLE_LOG_TO_STDOUT=0
set K4A_LOG_LEVEL=w
```

wes-b, 2019-09-17:
We can reopen this if you are still having the issue and can share the requested log

closed 2019-09-17 by wes-b