microsoft/Azure-Kinect-Sensor-SDK #723: Cmake Error in windows 10

wodndb, 2019-09-05:
|  |  |
|--|--|
| OS | Windows |
| Installed IDE | VS 2017, VS 2019 |
| Cmake version | 3.15.3 |
| Ninja version | 1.9.0 |
| Kinect V4 SDK version | 1.2.0 |

I installed SDK zip file and unzipped SDK to `Documents` folder.
I made `build` directory at the root directory of SDK and ran the command `cmake .. -GNinja` at `build`.
However I got cmake error message like this (↓).

```
-- Setting build type to 'RelWithDebInfo' as none was specified.
-- The C compiler identification is MSVC 19.16.27032.1
-- The CXX compiler identification is MSVC 19.16.27032.1
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe -- broken
CMake Error at C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Users/wodnd/Documents/Azure-Kinect-Sensor-SDK-1.2.0/build/CMakeFiles/CMakeTmp

    Run Build Command(s):C:/Windows/System32/ninja.exe cmTC_04871 && [1/2] Building C object CMakeFiles\cmTC_04871.dir\testCCompiler.c.obj
    [2/2] Linking C executable cmTC_04871.exe
    FAILED: cmTC_04871.exe
    cmd.exe /C "cd . && "C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_04871.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100177~1.0\x86\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100177~1.0\x86\mt.exe --manifests  -- C:\PROGRA~2\MIB055~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx86\x86\link.exe /nologo CMakeFiles\cmTC_04871.dir\testCCompiler.c.obj  /out:cmTC_04871.exe /implib:cmTC_04871.lib /pdb:cmTC_04871.pdb /version:0.0  /machine:X86  /debug /INCREMENTAL /subsystem:console  kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib && cd ."
    LINK Pass 1: command "C:\PROGRA~2\MIB055~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx86\x86\link.exe /nologo CMakeFiles\cmTC_04871.dir\testCCompiler.c.obj /out:cmTC_04871.exe /implib:cmTC_04871.lib /pdb:cmTC_04871.pdb /version:0.0 /machine:X86 /debug /INCREMENTAL /subsystem:console kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:CMakeFiles\cmTC_04871.dir/intermediate.manifest CMakeFiles\cmTC_04871.dir/manifest.res" failed (exit code 1104) with the following output:
    LINK : fatal error LNK1104: 'ucrtd.lib' 파일을 열 수 없습니다.
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:23 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/wodnd/Documents/Azure-Kinect-Sensor-SDK-1.2.0/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/wodnd/Documents/Azure-Kinect-Sensor-SDK-1.2.0/build/CMakeFiles/CMakeError.log".
```

Could you help me? I want to build SDK and test CSharp wrapper.

I attached CMakeError.log and CMakeOutput.log
[CMakeOutput.log](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/3577621/CMakeOutput.log)
[CMakeError.log](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/3577622/CMakeError.log)

wodndb, 2019-09-05:
I also tried [this solution](https://stackoverflow.com/questions/14590947/cmake-configuring-fails-cl-exe-is-not-able-to-compile-a-simple-test-program) but failed.

RoseFlunder, 2019-09-05:
Did you read all the steps in building manual?
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md

Download via zip file is not enough:

>  If you need to build from source, you have to clone the repository instead of downloading the Zip file to ensure all the submodule dependencies in place.

wodndb, 2019-09-06:
I installed all dependencies.
[image]
And also I use Native tool x64 prompt for Visual Studio 2017
However, I got same error message.

I think that VS 2017 has something wrong. I will try to reinstall VS 2017.

wodndb, 2019-09-06:
I fail to build although reinstall VS 2017 and cmake.

wodndb, 2019-09-06:
build works well based on Developer command prompt for VS 2019 (I have only x86 version although  Windows is x64)

wes-b, 2019-09-06:
From a command line launch the developer command prompt with ```-arch=amd64```. In my case it is:
```
"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\Tools\VsDevCmd.bat" -arch=amd64
```

Last paragraph here: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md#windows-dependencies.

wodndb, 2019-09-09:
@wes-b Thanks! I can run the command `cmake .. -GNinja` .However can't run the next build command `ninja`.

```
[562/740] Building CXX object src\logging\CMakeFiles\k4a_logging.dir\logging.cpp.obj
ninja: build stopped: subcommand failed.
```

Also there are several warnings about encoding problem.

wes-b, 2019-09-10:
Are you still using the ZIP? We don't support building the zip. You need to be building from a `Git clone` of the repro. As @RoseFlunder pointed out, please go through our [building](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) steps. We are happy to make changes if you encounter anything that seems unclear or confusing (feel free to file more issues). If you still run into problems please reiterate the steps you took and the results you got.

wodndb, 2019-09-11:
@wes-b I cloned repository using git because @RoseFlunder feedback to me.

wes-b, 2019-09-11:
Ok, so opening a developer VsDevCmd instance is working, and calling `cmake .. -GNinja` is working. But `ninja` is giving you a build error. Can you share the full build output?

wes-b, 2019-09-17:
We can reopen this if you are still having issues and can share the updated full build log

closed 2019-09-17 by wes-b

wodndb, 2019-09-18:
@wes-b I'm sorry. It's too late to share full build output of ninja. I had national holiday in South Korea (Chuseok)

```
C:\Users\USER\Documents\Azure-Kinect-Sensor-SDK\build>ninja
[0/2] Re-checking globbed directories...
[448/740] Building C object extern\libsoundio\src\CMakeFiles\libsoundio_static.dir\src\util.c.obj
cl : 명령줄 warning D9025 : '/W3'을(를) '/W4'(으)로 재정의합니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\util.h(89): warning C4244: 'return': 'double' 에서 'int'(으)로 변환하면서 데이터가 손실될 수 있습니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\util.h(93): warning C4244: '초기화 중': '__int64'에서 'double'(으)로 변환하면서 데이터가 손실될 수 있습니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\util.h(93): warning C4244: '초기화 중': '__int64'에서 'const double'(으)로 변환하면서 데이터가 손실될 수 있습니다.
..\extern\libsoundio\src\src\util.c(36): warning C4189: 'len2': 지역 변수가 초기화되었으나 참조되지 않았습니다.
[466/740] Building C object extern\libsoundio\src\CMakeFiles\libsoundio_static.dir\src\os.c.obj
cl : 명령줄 warning D9025 : '/W3'을(를) '/W4'(으)로 재정의합니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\util.h(89): warning C4244: 'return': 'double' 에서 'int'(으)로 변환하면서 데이터가 손실될 수 있습니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\util.h(93): warning C4244: '초기화 중': '__int64'에서 'double'(으)로 변환하면서 데이터가 손실될 수 있습니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\util.h(93): warning C4244: '초기화 중': '__int64'에서 'const double'(으)로 변환하면서 데이터가 손실될 수 있습니다.
..\extern\libsoundio\src\src\os.c(169): warning C4189: 'err': 지역 변수가 초기화되었으나 참조되지 않았습니다.
..\extern\libsoundio\src\src\os.c(268): warning C4189: 'ok': 지역 변수가 초기화되었으나 참조되지 않았습니다.
..\extern\libsoundio\src\src\os.c(266): warning C4189: 'err': 지역 변수가 초기화되었으나 참조되지 않았습니다.
..\extern\libsoundio\src\src\os.c(453): warning C4244: '초기화 중': 'double'에서 'DWORD'(으)로 변환하면서 데이터가 손실 될 수 있습니다.
..\extern\libsoundio\src\src\os.c(613): warning C4244: '초기화 중': 'size_t'에서 'double'(으)로 변환하면서 데이터가 손실될 수 있습니다.
..\extern\libsoundio\src\src\os.c(613): warning C4244: '초기화 중': 'size_t'에서 'const double'(으)로 변환하면서 데이터 가 손실될 수 있습니다.
..\extern\libsoundio\src\src\os.c(614): warning C4244: 'return': 'double'에서 'size_t'(으)로 변환하면서 데이터가 손실될 수 있습니다.
..\extern\libsoundio\src\src\os.c(622): warning C4267: '인수': 'size_t'에서 'DWORD'(으)로 변환하면서 데이터가 손실될 수 있습니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\os.c(588) : warning C4706: 조건식 내에 할당이 있습니다.
[497/740] Building C object extern\libsoundio\src\CMakeFiles\libsoundio_static.dir\src\ring_buffer.c.obj
cl : 명령줄 warning D9025 : '/W3'을(를) '/W4'(으)로 재정의합니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\util.h(89): warning C4244: 'return': 'double' 에서 'int'(으)로 변환하면서 데이터가 손실될 수 있습니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\util.h(93): warning C4244: '초기화 중': '__int64'에서 'double'(으)로 변환하면서 데이터가 손실될 수 있습니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\util.h(93): warning C4244: '초기화 중': '__int64'에서 'const double'(으)로 변환하면서 데이터가 손실될 수 있습니다.
..\extern\libsoundio\src\src\ring_buffer.c(14): warning C4100: 'soundio': 참조되지 않은 정식 매개 변수입니다.
..\extern\libsoundio\src\src\ring_buffer.c(91): warning C4267: '=': 'size_t'에서 'int'(으)로 변환하면서 데이터가 손실될 수 있습니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\ring_buffer.c(87) : warning C4706: 조건식 내에 할당이 있습니다.
[499/740] Building C object extern\libsoundio\src\CMakeFiles\libsoundio_static.dir\src\soundio.c.obj
cl : 명령줄 warning D9025 : '/W3'을(를) '/W4'(으)로 재정의합니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\util.h(89): warning C4244: 'return': 'double' 에서 'int'(으)로 변환하면서 데이터가 손실될 수 있습니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\util.h(93): warning C4244: '초기화 중': '__int64'에서 'double'(으)로 변환하면서 데이터가 손실될 수 있습니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\util.h(93): warning C4244: '초기화 중': '__int64'에서 'const double'(으)로 변환하면서 데이터가 손실될 수 있습니다.
..\extern\libsoundio\src\src\soundio.c(173): warning C4100: 'soundio': 참조되지 않은 정식 매개 변수입니다.
..\extern\libsoundio\src\src\soundio.c(174): warning C4100: 'msg': 참조되지 않은 정식 매개 변수입니다.
..\extern\libsoundio\src\src\soundio.c(176): warning C4100: 'soundio': 참조되지 않은 정식 매개 변수입니다.
..\extern\libsoundio\src\src\soundio.c(460): warning C4100: 'os': 참조되지 않은 정식 매개 변수입니다.
..\extern\libsoundio\src\src\soundio.c(464): warning C4100: 'outstream': 참조되지 않은 정식 매개 변수입니다.
..\extern\libsoundio\src\src\soundio.c(568): warning C4244: '인수': 'double'에서 'float'(으)로 변환하면서 데이터가 손실 될 수 있습니다.
..\extern\libsoundio\src\src\soundio.c(571): warning C4100: 'is': 참조되지 않은 정식 매개 변수입니다.
..\extern\libsoundio\src\src\soundio.c(575): warning C4100: 'instream': 참조되지 않은 정식 매개 변수입니다.
..\extern\libsoundio\src\src\soundio.c(704): warning C4100: 'soundio': 참조되지 않은 정식 매개 변수입니다.
..\extern\libsoundio\src\src\soundio.c(708): warning C4100: 'soundio': 참조되지 않은 정식 매개 변수입니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\soundio.c(191) : warning C4706: 조건식 내에 할당이 있습니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libsoundio\src\src\soundio.c(237) : warning C4706: 조건식 내에 할당이 있습니다.
[501/740] Building C object extern\libsoundio\src\CMakeFiles\libsoundio_static.dir\src\dummy.c.obj
cl : 명령줄 warning D9025 : '/W3'을(를) '/W4'(으)로 재정의합니다.
[... cut ...] c\record\internal\matroska_read.cpp
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libebml\src\ebml\EbmlEndian.h: error C2220: 경고가 오류로 처리되 어 생성된 'object' 파일이 없습니다.
c:\users\user\documents\azure-kinect-sensor-sdk\extern\libebml\src\ebml\EbmlEndian.h: warning C4819: 현재 코드 페이지(949)에서 표시할 수 없는 문자가 파일에 들어 있습니다. 데이터가 손실되지 않게 하려면 해당 파일을 유니코드 형식으로 저장하십 시오.
[581/740] Building CXX object tests\ColorTests\UnitTest\CMakeFiles\color_ut.dir\color_ut.cpp.obj
ninja: build stopped: subcommand failed.
```

wes-b, 2019-09-18:
Can you work around 'warning C4819' in 'EbmlEndian.h' by:
>To resolve C4819, save the file in Unicode format. In Visual Studio, choose File, Advanced Save Options. In the Advanced Save Options dialog box, select an encoding that can represent all the characters in the file—for example, UTF-8—and then choose OK.

wodndb, 2019-09-19:
@wes-b I have not enough time to work. I will try it if I have a chance to get spare time.

ishanivyas, 2019-11-02:
Are you able to solve the problem? I am having the same issue on window10.