microsoft/Azure-Kinect-Sensor-SDK #725: How can I solve the problem of the depth and color images quality gotten by k4a_transformation functions ?

nonlinear1, 2019-09-05:
Dear
For k4a_transformation functions of SDK, I have used the funciton k4a_transformation_depth_image_to_color_camera to get registered image pairs of color and depth. Both are 1024*768 resolution, at 30fps. I use this configuration to run orbslam2 of rgbd camera version. I found that orbslam2 with kinect for azure often tracked lost compared to other rgbd camera. After careful checking up,I found that the one third of the depth image has zero values at image edge(about one-sixth at left and one-sixth at right, so in all one third of the depth image) and all the feature points in this area cannot become map points. And If I cut the one third of depth image, which has zero values, I think the angle of view of depth image is narrow and this also can cause tracking lost easily.
If I use k4a_transformation_color_image_to_depth_camera to get registered image pairs of colour and depth. And both are 640*435 resolution, at 30fps. If I use this configuration to run orbslam2. I found that it almost cannot work for the colour image has bad quality, maybe colour image has many holes. Now, I want to get 640*435 resolution registered image pairs and the image quality is good.
Could someone know how can I get the 640*435 resolution registered good quality image pairs? And any good suggestion to solve the problem of orbslam2 tracking lost easily?

Thank you very much

rabbitdaxi, 2019-09-05:
@nonlinear1 , can you be more specific on which depth camera mode you have used? 
Please read the [doc](https://docs.microsoft.com/en-us/azure/Kinect-dk/use-image-transformation) , which explains how the transformation method work. Please also read [this](https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification), which explains the hardware specification regarding color depth camera FOV compare. 

Regarding RGBD SLAM, I have some feedback hopefully helps you:
- First of all, here is one open sourced RGBD SLAM which naively work with Azure Kinect: https://github.com/ETH3D/badslam, you might can take it as a reference
- If you use both color and depth camera, then it really depends on the overlapped FOV. And based on the hardware specification spec, you can try various combinations of color and depth mode, however, even the depth camera can go to 120x120, the color camera is not. 
- 30fps plus moving camera slowly can help you have more frame to frame correspondence that orbslam2 highly depends on
- If you use depth camera's depth + ir image, the fov can go to 120x120 which may help, but ir image could mean less features because the signal decrease over distance
- The device has IMU, it might be a good project to leverage IMU to make tracking robust and avoid tracking lose

nonlinear1, 2019-09-05:
Thank you for your help, again. I have tested badslam. It is very good. But If I run badslam, it need GPU. For now, I want run a slam on computer without GPU.
    the configuration for azure kinect I used to run orbslam2 is as follow:
  k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
    config.synchronized_images_only = true;
    config.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;//examples/transformation
    config.color_resolution = K4A_COLOR_RESOLUTION_720P;
    //config.depth_format = K4A_IMAGE_FORMAT_DEPTH16;//https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/
    config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;//640x576.
    config.camera_fps = K4A_FRAMES_PER_SECOND_30;

also I have used the funciton: 
k4a_transformation_depth_image_to_color_camera(transformation, depth_image, transformed_depth_image)
get registered image pairs of color and depth image. both are 1024*768 resolution.

nonlinear1, 2019-09-05:
the input file(K4A.cpp) I used to run orbslam2 with kinect for azure is as follow:

#include <iostream>
#include <k4a/k4a.h>
#include <opencv2/core/core.hpp>
#include <opencv2/highgui/highgui.hpp>
#include <opencv2/imgproc/imgproc.hpp>
#include <System.h>   // orb_slam2

using namespace std;
using namespace cv;

template<typename T> Mat create_mat_from_buffer(T *data, int width, int height, int channels = 1)
{
    Mat mat(height, width, CV_MAKETYPE(DataType<T>::type, channels));
    memcpy(mat.data, data, width * height * channels * sizeof(T));
    return mat;
}

static string get_serial(k4a_device_t device)
{
    size_t serial_number_length = 0;

    if (K4A_BUFFER_RESULT_TOO_SMALL != k4a_device_get_serialnum(device, NULL, &serial_number_length))
    {
        cout << "Failed to get serial number length" << endl;
        k4a_device_close(device);
        exit(-1);
    }

    char *serial_number = new (std::nothrow) char[serial_number_length];
    if (serial_number == NULL)
    {
        cout << "Failed to allocate memory for serial number (" << serial_number_length << " bytes)" << endl;
        k4a_device_close(device);
        exit(-1);
    }

    if (K4A_BUFFER_RESULT_SUCCEEDED != k4a_device_get_serialnum(device, serial_number, &serial_number_length))
    {
        cout << "Failed to get serial number" << endl;
        delete[] serial_number;
        serial_number = NULL;
        k4a_device_close(device);
        exit(-1);
    }

    string s(serial_number);
    delete[] serial_number;
    serial_number = NULL;
    return s;
}

int main(int argc, char **argv)
{ 
    if(argc != 3)
    {
       cerr << endl << "Usage: ./rgbd_cc path_to_vocabulary path_to_settings" << endl;
       return 1;

    }
    // construct ORB_SLAM system
    ORB_SLAM2::System SLAM(argv[1],argv[2],ORB_SLAM2::System::RGBD,true);
 
    cout << endl << "-------" << endl;
    cout << "Openning Kinect for Azure ..." << endl;
 
    cv::Mat imBGRA, imBGR, imD, imIr, im_transformed_color_image;
    cv::Mat im_transformed_depth_image;
   
[... 154 lines cut ...]
                return false;
           }
      
           //std::cout << "width = " << width << std::endl;
           //std::cout << "height = " << height << std::endl;
           cv::Mat frame;
           uint8_t *buffer = k4a_image_get_buffer(transformed_depth_image);
           uint16_t *depth_buffer = reinterpret_cast<uint16_t *>(buffer);
           create_mat_from_buffer<uint16_t>(depth_buffer, color_image_width_pixels, color_image_height_pixels).copyTo(frame);
            
           if(frame.empty())
           cout << "im_transformed_color_image is empty" << endl;
          // imshow("transformed_depth_image", frame);
           //waitKey(0);
           SLAM.TrackRGBD( imBGR, frame,  index);
           k4a_image_release(depth_image); 
           k4a_image_release(color_image); 
           k4a_image_release(transformed_depth_image); 
           k4a_capture_release(capture);
    
    }
     SLAM.Shutdown();
     return 0;
    
}

nonlinear1, 2019-09-06:
I have used kinect for azure and orbbec astro mini (a rgbd camera the resolution is 640*480 for both image and color image) to run the orbslam2 with same parameters configuration for orbslam2 for comparation in the same place.The Azure kinect  easily tracking lost with less initial mappoints with 371 mappoints and only in the middle of the image. To the contrast, The Astro mini tracking good with more initial mappoints with 692 mappoints and the mappoints spreads all the image equally. I think maybe this is because I used the function "k4a_transformation_depth_image_to_color_camera" to get registered pairs of color and depth image. the original depth resolution is 640*576 and after using this function it become 1280*720. I think maybe the depth value is not exact, especially in the edge of depth image. About one-third of the transformed depth image has zero depth (about one-sixth at left and one-sixth at right, so in all one third of the depth image). Could you have suggestion?@rabbitdaxi

nonlinear1, 2019-09-06:
this is the initial mappoints image of orbslam2 with kinect for azure:
[image]

nonlinear1, 2019-09-06:
the following is the initial mappoints image of orbslam2 with orbbec astro mini
[image]

nonlinear1, 2019-09-06:
@rabbitdaxi Do you have email? I think putting the compared results in here maybe not good. Maybe I should send them to you.

rfilkov, 2019-09-06:
@nonlinear1 Why do you use a NFOV depth mode? It has Narrow field of view, as you can see on your screenshot. You better try with one of the WFOV-depth modes (2x2 binned @ 30 fps or unbinned @ 15 fps). They will cover the full color camera FOV. The tracked distance may be an issue there, but I suppose the result will be better than with the Astra sensors.

nonlinear1, 2019-09-06:
I have tested 2x2 binned @ 30 fps to run orbslam2 with Kinect for Azure and Astra mini.The same result. Track lost more easily than Astra mini. And have less initial mappoints than Astra mini

nonlinear1, 2019-09-06:
I don't know why 2x2 binned @ 30 fps mode should good depth image quality for the depth values in the edge have not been zeros. Why the same results? the depth value in the feature point isn't exact?
For feature points may be at the place, where the depth value is discontinuous. So when I use the k4a_transformation_depth_image_to_color_camera function. the depth image is zoomed in. I think some depth value in zoomed in depth is interpolated. Could someone have any suggestions for improving the tracking quality of orbslam2 with kinect for azure?Thank you very much

rfilkov, 2019-09-06:
Could you please post a screenshot again, with WFOV-depth mode this time? This issue with WFOV modes is that they should be used for relatively near distances - within ~3 meters, I think.

nonlinear1, 2019-09-06:
@rfilkov OK, this time I change the contrast and sharpness of the kinect for azure , then run orbslam2. the sharpness and contrast configurarion is as follow:
 k4a_device_set_color_control(device, K4A_COLOR_CONTROL_CONTRAST, K4A_COLOR_CONTROL_MODE_MANUAL, 7);
    k4a_device_set_color_control(device, K4A_COLOR_CONTROL_SHARPNESS, K4A_COLOR_CONTROL_MODE_MANUAL, 4);
this time I got more initial mappoints than orbbec. and for tracking quality is orbbec good and kinect for azure is easily tracking lost. I found the mappoints distribution is very different!!! The compared images are as follows:
First,I use kinect for azure to run orbslam2.
[image]
second,I use the orbbec astro mini to orbslam2 in the same palce and almost the same time
[image]
I found the mappoints or the feature points(I have extracted featrues with the way of orbslam2) distributed in the egde of square for orbbect astro mini. But the mappoints or feature points of kinect for azure spread randomly. I think the feature points of kinect for azure is unstable? Maybe this is the reason why I use kinect for azure to run orbslam2 easily tracking lost.Could you some have any suggestion?

rabbitdaxi, 2019-09-06:
@rfilkov Thank you for providing insights and we appreciates the community to discuss and solving issue together on Github. 

@nonlinear1 I just quickly read your many posts this morning. Thank you for the continuous effort. I don't have bandwidth to debug and run orbslam2, however, based on your images and discussion above, here are some my personal thoughts hopefully can help:
- I agree with @rfilkov that you should try out WFOV as well, particularly I think the WFOV_2x2Binned should be a good candidate. The color camera fov for the more you are chosen is 90°x59°, and depth camera's fov for NFOV is 75°x65°, WFOV is 120°x120°. Therefore, when you transform the depth image to color camera, if you want to have the field view cover the color camera, definitely the WFOV mode is a good candidate compare to NFOV. You may see less range though compare to NFOV, but worth to try.
- I played with orbslam2 a while back, not particularly on this camera, but for my own research project, I remember may be orbslam2 does have filtering the visual features if the depth is invalid (here with 0 value in Azure kinetic)? If this is still the case, you might want to debug to see how many visual features are eliminated because the transformed depth pixel is invalid. This camera is a time-of-flight camera, with all the benefits of low jitter, low power, high accuracy, sometimes, the invalidation of some bad pixels need to happen. See [page](https://docs.microsoft.com/en-us/azure/Kinect-dk/depth-camera). I did not use the orbbec camera before, sometimes when you compare depth, you need to compare point cloud quality instead of just look at 2d depth image. 
- Now, one thing you might want to dig into is, have you tried just run mono with the color image only and see whether it is still the case? In this way, you can narrow down whether the issue is related to transformed depth, or it is mostly coming from the feature points based on the color image?
- By the way, BAD-SLAM is using the same depth to color transformation method that you are doing. I can try forward this github issue to someone did that work. If they have time, they may have more insights.

nonlinear1, 2019-09-07:
@rabbitdaxi Thank you very much, This week I will run mono version of orbslam2 with color image.
Last night, I think maybe the color camera of kinect for azure is more resolution than orbbec(resolution:640x480).So, at the same place, the color camera of the kinect can catch more details than orbbec, And these details are extracted as feature points, and these subtle details of object  are unstable and susceptible to illumination change. As the above image, I have posted, the color image of orbbec is more blurry than kinect for azure. I think maybe I should blur the color image of kinect for azure with Gaussian blur，Firstly. Then run orbslam2, this maybe spotlight the main featrue points. Do you think so? or have any suggestion?
    And several week ago, I have run BAD-SLAM, successfully, give some issue about BAD-SLAM, your colleague have give some explanations. When I run BAD-SLAM, it more easily track lost than orbslma2 with kinect for Azure. But, I think the BAD-SLAM is more better than other slam. I really like it. Maybe, I will spend more time to regulate some parameters.

nonlinear1, 2019-09-09:
I have run mono version of orbslam2 with color image. I cann't find differences between orbbec rgbd camera and kinect for azure. And Kinect for Azure has more initial mappoints.
I have use the bilateralFilter to process the color image. Then as I gussed above, I can get the similar feature points distribution with orbbec. But I cann't run bilateralFilter realtime with cpu. It is too much computation. Here is my compared images of extracted orb features with the same way of orbslam .
First, orbbec
[image]
Second, kinect for Azure without bilateralFilter
[image]
Thrid, kinec for Azure with bilateralFilter
[image]

nonlinear1, 2019-09-09:
@rabbitdaxi and @rfilkov So, with bilateralFilter I can get the similar orb features distribution. If I can run orbslam2 with bilateralFilter to process the color image and WFOV mode(512*512 resolution) in realtime. I think I can get the same results with orbbec or may better results than orbbec. But run bilateralFilter in cpu is too different in realtime! any suggestions?

kysucix, 2019-09-09:
hi @nonlinear1 ! First of all the problem of missing features on the corners is only due to the different field of view of RGB and Depth stream.
See this picture extracted from the specification of the kinect https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification#camera-field-of-view:
[image]

Then I guess the reason for the difference in the amount of extracted features is the higher resolution of the kinect rgb camera: when using higher resolution input you need to adjust the number of scales used to compute the orb feature, to compensate for the resolution difference.
When you're filtering the image with bilateral filter (likely also a gaussian would work) you're practically reducing the resolution of the input image, matching the "standard" one used by orb-slam.
You're not the first to experience these kind of problems on higher resolution input:
https://github.com/raulmur/ORB_SLAM2/issues/35

Btw don't forget to remove the lens distortion (from both rgb and depth) before passing the image to orbslam.
Have a look at the source code of badslam to see how it's done:
https://github.com/ETH3D/badslam/blob/master/applications/badslam/src/badslam/input_azurekinect.cc

nonlinear1, 2019-09-09:
@kysucix Thank you very much for your detailed explanation！

wes-b, 2019-09-09:
Sounds like we got this issue resolved, please ping if not.

closed 2019-09-09 by wes-b