microsoft/Azure-Kinect-Sensor-SDK #726: Manual / user-defined intrinsic & extrinsic calibration?

fishcu, 2019-09-05:
Thanks for creating this SDK (mostly) open source. It's very useful already.

One thing I noticed today: It's not possible to replace the factory camera calibration by another calibration that is created manually.

I haven't yet tested how good the factory calibration is. It seems to be pretty good, judging only visually, but maybe not perfect. Overlaying depth and color shows some slight misalignments towards the edges of the image frame (where effects of distortion are the strongest).

Do you plan to add any functionality for the user to supply a custom calibration?

Looking at other products like stereolab's ZED camera, you can see that they provide a calibration tool with user feedback, which handles storage and calculation of a device's intrinsics & stereo extrinsics automatically.

Storing the calibration could be done in a few ways:
- Provide a file format and function with which you can load from / store to the disk. This would replace getting a calibration from a device handle in such applications
- Provide setter-functions so that the custom calibration can be stored on the device. Don't know if this is supported by the hardware. The factory calibration would remain available at all times, should the user screw up the calibration somehow.

Calculation could be done by:
- Creating functions which can calculate intrinsics & extrinsics from calibration target & measured  points. There could be an example / tool built around these functions which display a checkerboard on screen, uses OpenCV to detect the checkerboard, and can calibrate automatically (with some user feedback)
- Using OpenCV, such as in other examples, to detect a calibration target (checkerboard), but also to estimate distortion parameters. The distortion model (Brown-Conrady) used seems to be compatible with OpenCV currently anyway. IIRC OpenCV would also support a rational model, but I don't know if it aligns with Azure Kinect SDK's model.

rabbitdaxi, 2019-09-09:
@fishcu  thank you for the feedback. I can add few comments on your observation of the depth color overlay at edges. 
- There are some [doc](https://docs.microsoft.com/en-us/azure/Kinect-dk/use-image-transformation) here describe the functions and also the reference code for these transformation functions are open sources in the Github. 
- Recommend to use the depth to color camera transformation since it handles occlusion. Due to performance reason, the color to depth transformation does not handle occlusion. 
- If the slight misalignment on the edge you are talking about is not from occlusion, then please keep in mind there are multiple reasons can lead to this slightly misalignment. e.g. the resolution of the color and depth are quite different, therefore, during transform, due to resolution difference, slightly imperfect edge in depth can be amplified in color; calibration accuracy is one possible reason too; other than that, depth accuracy, even it is quite small, but can contribute to this as well since we are using depth value to transform. 

Currently there is no plan to add support of overriding the factory calibration. You still can do your own calibration and just create the k4a_calibration_t object from your own calibration file format. The object type is defined in https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/structk4a__calibration__t.html. One [example](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/structk4a__calibration__t.html), around L230 ish, since it create a new extrinsic calibration between two devices, and to help the transformation, it create a new "virtual" calibration object by using the manual calibrated extrinsics and the factory intrinsics, then use it later in the code. 

Please let us know whether this can unblock you.

fishcu, 2019-09-11:
Thank you for the useful summary. 

I have one more question: Is it possible that not all distortion models are implemented yet? I could only find the relevant code for the standard Brown and `RATIONAL_6KT` models. I have not tried undistorting the WFOV depth yet, but I can imagine that these models might not be the best fit because of the fisheye lens. I'm guessing that `K4A_CALIBRATION_LENS_DISTORTION_MODEL_THETA` would correspond better to the fisheye model.

rabbitdaxi, 2019-09-11:
@fishcu [Documentation](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___enumerations_ga1c33fe57cf4e99fcfd556be60fce4d35.html#ga1c33fe57cf4e99fcfd556be60fce4d35) quote "Azure Kinect devices are calibrated with Brown Conrady." The lens is not a fisheye lens, it is a normal lens with wide field of view. The team did the calibration have verified rational6kt or brownconrady is the best fit for this camera. To be compatible with opencv, browncontrady was chosen for the final product factory calibration. 

Yes, there are some enum value in the calibration model type (theta/polynomial) was not implemented (they are legacy code we should clean up, I will send a PR for that)

jasjuang, 2019-11-23:
I think it will be a good idea to have the functionality to override factory calibration because of #803.

interwebjill, 2020-06-16:
@rabbitdaxi
In your comment of  9 Sep 2019, you referenced an example. The link is incorrect. I would like to review this example. Could you please provide the link you intended?

rabbitdaxi, 2020-06-16:
@interwebjill corrected the link inline in the comment I made. Hope it helps