microsoft/Azure-Kinect-Sensor-SDK #733: ffmeg extracted Passive IR Images are Black while Viewer passive IR images look good

xanderdunn, 2019-09-06:
In the Azure Kinect Viewer I see passive IR images as I expect to see them:
[image]

However, in the same conditions, when using the Azure Kinect Recorder and then ffmpeg to output to .png as described [here](https://docs.microsoft.com/en-us/azure/kinect-dk/record-file-format), I get almost completely black images:
[image]

Even though the output of the RGB camera at the same frame looks fine:
[image]

### Reproduce:
- Collect data with `"C:\Program Files\Azure Kinect SDK v1.2.0\tools\k4arecorder.exe" --depth-mode PASSIVE_IR --color-mode 1080p --imu OFF --rate 15 C:\Users\xander\Desktop\output.mkv`
- Using the same collection settings, look at the passive IR image in the Azure Kinect Viewer
- The two sets of results are remarkably different but I'd expect them to be the same.

### Environment
Windows 10.0.17763 Build 17763
Azure Kinect SDK Version 1.2.0
Azure Kinect Firmware:
```
  RGB camera firmware:      1.6.98
  Depth camera firmware:    1.6.70
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft
```

rabbitdaxi, 2019-09-06:
@xanderdunn passive IR mode means there is no active illumination from the depth camera. The reason the Azure Kinect viewer looks bright might because the scale? By searching in the source code of the k4aviewer.exe, I found this function `GetIrLevels` in this [file](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/tools/k4aviewer/k4astaticimageproperties.h). And I think what is happening is the passive IR image contains very small values, and the k4aviewer did a min/max scale with [0 - 100], therefore, with the normalization, even values less than 100 looks bright. But since the ir image is uint16_t, when you use the k4arecorder and ffmpeg, it stores the true value in the 16bit image, which appears very dark.

xthexder, 2019-09-11:
The images output by ffmpeg are 16-bit grayscale png's. These are tricky to display, since when scaled to 8-bit color for your monitor, most of the data is displayed as black/near-block.
You can actually see one spot in your image where the IR is saturated, making it visible even when scaled to the 8-bit range.

To verify the PNGs actually contain the data you need, you can use an image editor such as [GIMP](https://www.gimp.org/) to adjust the levels.
You'll want to change the input range max value:
[image]

I believe ffmpeg is able to do this directly using some of the video filters available, but I'm not sure if there's any that support a 16-bit color space.

rabbitdaxi, 2019-09-11:
@xanderdunn please let us know whether the answers above solved your issue. If so, we can close this issue.

rabbitdaxi, 2019-09-13:
@xanderdunn Based on the initial post we believe the answers above should explain what you see, you just need to look at the images from ffmpeg with the pixel inspector, it is just how the image scale results the dark image. In general, the value of the pixel intensity is expected to be small when you are in passive IR mode, given there is no active illumination.

Given have not heard from you more than 1 week, we close this issue for now, assuming the information above did help you. Let us know if you still have issues around this.

closed 2019-09-13 by rabbitdaxi

aleksjr98, 2023-06-04:
@xanderdunn @rabbitdaxi  how was this issue solved? How did you rescale the videos so they appear as they did in the previews?