microsoft/Azure-Kinect-Sensor-SDK #734: Confidence scores for Body Tracking Joints

hmexx, 2019-09-06:
Most 2D Pose Estimation models output a confidence score (0-1.0) for each joint, in addition to their XY position. This allows applications built on top of the model to gracefully handle situations like occluded joints.

Could we see this added to the Tracking SDK?

qm13, 2019-09-06:
Thanks for the feedback. The team has had intense debates on whether to expose a joint confidence. Assuming you had a joint confidence how would you make use of it in your application? How would it be reflected in the user experience?

RubensMoreno, 2019-09-06:
My team is also very interested on have confidence score or something similar for each joint not only for joint position or if it is occluded or not but also from the rotation perspective. For us is very important to know if we can trust or not in rotation data to determinate if we need to add some corrections to rotations for example.

RoseFlunder, 2019-09-06:
@qm13
We are trying to build setup a multi kinect system in a room with several things that can occlude parts of the bodies but when we use multiple cameras there are at least one or two that could see the body without occlusion. Our goal is to fuse the joint positions from the different kinects to get a fused skeleton. It would be very helpful to have a confidence per joint per Kinect for that which will affect the weighting for this joints position in the data fusion

rfilkov, 2019-09-06:
@qm13 here is an use case: I'd like to overlay the user's joint with virtual object. Let's say I want to put a ball of fire in the user's hand on screen. Let's forget for a moment that hands are not tracked (yet). If the user moves his/her hand behind the body, the ball of fire should disappear. It would be easy to do that, if there were tracking states or confidence scores. Without this feature, one should guess whether the hand is visible or not.

billpottle, 2019-09-08:
@qm13 - This could also help provide a workaround for issues such as #738 where developers could filter out obvious inaccurate positions by some combination of the confidence score and velocity (ie, joint position 'moved' 1 meter within 1 frame. )

hmexx, 2019-09-10:
Pretty much every application of pose estimation would benefit from joint confidence scores. Even a simple skeletal visualization looks odd at the moment because of the inhuman poses the SDK estimates when joints are occluded, instead of just hiding them. Another example is form analysis of an athlete during an exercise. If a joint has low confidence, we'd prefer to ask the athlete to change positions (remove occlusions) rather than present nonsensical feedback to them.

Hope something can be put together soon, as we are having to explore alternative models that use the kinect depth data. On that note, hope it's not too rude to ask the community here whether they have found any good alternatives? For instances, has anyone tried this one: https://github.com/lmb-freiburg/rgbd-pose3d @billpottle @rfilkov @RoseFlunder 

Thanks

RoseFlunder, 2019-09-10:
@hmexx I haven't looked for alternatives that use depth data yet. The current system still uses the kinect v2 which has a confidence value. At the moment I am still working on replacing the kinects v2 with azure kinects and integrate them in our ROS environment. Therefore I did the some work on the azure kinect ROS package (i.e. added body tracking data publishing). However a colleague mentioned we could compare the results to open pose, which doens't use depth data, but has a 3D pose estimation for multi view (we use four kinects).

qm13, 2019-09-24:
The team has debated internally for some time on whether to expose some form of joint confidence. We are pleased many in the community have asked for joint confidence and explained your use cases. Thank you.

We are proposing to provide a joint confidence indicator for each joint. The indicator has one of four values:

CONFIDENCE_HIGH - high confidence in joint pose
CONFIDENCE_MEDIUM - medium confidence in joint pose
CONFIDENCE_LOW - not observed (likely due to occlusion), predicted joint pose
UNSUPPORTED - joint out of range (too far from depth camera).

We are specifically NOT exposing a joint numerical weight (0 - 1). Weights are not consistent between model adjustments and training. We believe relying on a numerical weight will lead to backward compatibility problems.

We look forward to your feedback. 

Regards
Azure KInect Body Tracking team

RoseFlunder, 2019-09-24:
Thank you for the update. This will be very helpful in my application.

hmexx, 2019-09-24:
A 3/4 tier confidence score will probably be sufficient for a majority of
use cases, although I m sure there are some cases where a full range (0-1)
might yield some small benefits as well.

I’m happy with the proposal in any case and look forward to seeing it
implemented. Any ideas on how much work it is and how long it will take?

Thanks for listening.

PierrePlantard, 2019-09-24:
Sound great for us too.
Thank you @qm13 for the feedback.
It could be useful to explain the differences between High and Medium confidence (Low = not visible, it's clear).
Regards

Brekel, 2019-09-24:
Such an indicator would indeed perfectly fit my needs. Thanks for listening to our feedback!

rfilkov, 2019-09-24:
Sounds great! It would be fairly easy to convert these levels to confidence score. As to me, only the lowest confidence level should have a consistent name, something like CONFIDENCE_NONE instead of UNSUPPORTED. But this is a cosmetic issue.

@qm13 Could you please explain (maybe with a simple example), what would be the difference between the high and medium joint confidence? The other two levels are pretty clear.

qm13, 2019-09-24:
Thank you for the prompt feedback. What is the difference between HIGH and MEDIUM you ask? We have an implementation of the API that we are currently evaluating against the key performance indicators we use to measure algorithmic improvement. Once this analysis is complete the intent is to provide the developer community with a defined statistical cut off between MEDIUM and HIGH. Stay tuned.

rbteyedeng, 2019-10-02:
@RoseFlunder @qm13 I am a researcher from the Microsoft skeleton tracking team. Internally, we developed the 3D skeleton fused system for multiple cameras using detection weights. Depending on my experiences, the confidence scores from 2D DNN output is not very dependable. So we come with another solution. The scores come out of  a combination of multiple factors, mostly influenced by 3D geometry. 

1. Each joint has a intrinsic given score from 0~1. joints such as shoulder, hip are more stable than ears, eyes thus getting  a higher score.

2. The relative orientation of the skeleton to the camera. If the camera see more of the side view of the body, the score is low and vice versa.

3.   The distance of the skeleton to the camera. closer--->higher score  

We are in the process of polishing the code and will release it if approved.

RoseFlunder, 2019-10-02:
@rbteyedeng
Thats sounds great and I want to follow the same approach, using distance + orientation to the camera as additional factors. So I am looking forward to your code.

But I the confidence indicator @qm13 mentioned will help a lot when some joints are just occluded for certain cameras and therefore the position is just kind of guessed.
In my scenario there will be multiple persons which can partly occlude each other for certain cameras.
Moreover there will be other things that can occlude joints like a bed or a wheelchair.
So I am planning to exclude joint positions (weighting factor 0) if the confidence value @qm13 proposed is UNSUPPORTED or CONFIDENCE_LOW.

qm13, 2019-11-25:
Released in v0.9.4

closed 2019-11-25 by qm13

laurentopia, 2020-01-20:
> We are specifically NOT exposing a joint numerical weight (0 - 1). Weights are not consistent between model adjustments and training. We believe relying on a numerical weight will lead to backward compatibility problems.

I'd like to offer a different perspective.

I've worked on a kinect1 multi sensor tracking which offers confidence 0-1.
An enum offers great help to make the decision to include a bone[sensor] but as a bone crosses the threshold between two acceptable enums MEDIUM and HIGH the resulting weighted average pops.
(using MEDIUM instead of clipping it out to smooth noise without having to cache frames)
A way to mitigate this is to lerp through time an internal weight mapped to the enum but an easier way and a more accurate way is to use your internal 0-1, the one you use to threshold into the enum, to do that weighted average.
Even if that 0-1 changes "curvature" as you change NN model and training methodology, this will help, besides, you can always provide, in the SDK, the internal weight value of each enum and we can fit a curve to it to more closely match how your NN model generates that bon[sensor].confidence.

I might be wrong in my methodology, in which case I welcome information in this domain.

Michael-robo13, 2020-05-29:
@qm13 @RoseFlunder I have a question about the body tracking sdk. Can it be used in raspberry pi 4? I want to obtain the skeleton data in system of raspberry pi4 + ros + azure kinect. Looking forward to your reply.

RoseFlunder, 2020-05-30:
@Michael-robo13 i am no member of the Microsoft team but as far as I know the Body Tracking SDK is dependant on a Deep Neural Network thats using CUDA. So you need a NVidia GPU which the Raspi does not have.