microsoft/Azure-Kinect-Sensor-SDK #738: Fully extended joints snap back to centerline for all smoothing values on 0.92

billpottle, 2019-09-06:
**Describe the bug**
When a distal joint is far from the center of the body, sometimes the whole limb will 'snap back' to the centerline, before correctly re-positioning itself as joints up the tree are flexed. 

Using k4abt_tracker_set_temporal_smoothing helps, but this bug persists across all smoothing values from 0 to 1. 

**To Reproduce**
Make a motion similar to what is shown in the attached gif at about 3/4 speed. 

**Expected behavior**
I would expect that foot position could rely on the current knee and ankle positions as well as their positions from the previous several frames to know that an extended position is much more likely than going back to the position of the other leg. 

**Screenshots**
See the attached animated gif. 

**Desktop (please complete the following information):**
 - OS with Version: Windows Version 10.0.18362 Build 18362
 - GPU: GTX 1080
 - SDK Version:  1.2 with body tracking 0.92

[image]

billpottle, 2019-09-21:
Update for 0.9.3 - The gif below is the same technique run through the new 0.9.3 version with smoothing set to 0. It seems like the problem may now be slightly worse, in addition to the standing leg being now located improperly. 

[image]

qm13, 2019-09-23:
Fixed in 0.9.3

billpottle, 2019-09-23:
@qm13- I am still having this issue in 0.9.3. Please see the above gif of the same MKV file run through the new SDK.

qm13, 2019-11-19:
@billpottle can you verify that this still happens in 0.9.4?

billpottle, 2019-11-21:
@qm13 - Yes, I can verify that it still happens on 0.9.4

qm13, 2019-12-03:
@billpottle we would like to get a recording that demonstrates the problems you are reporting. We require a signed data release agreement along with the recording. Could you provide me with contact details so I can provide you both the agreement and upload instructions?

billpottle, 2019-12-03:
@qm13 You can email me at bill@kattaekwondo.com

billpottle, 2019-12-11:
@qm13 - I haven't received the email with the data release agreement yet. You can also try myaicoach@hotmail.com

qm13, 2019-12-19:
@billpottle we have reproduced this bug and are actively working on a fix.