microsoft/Azure-Kinect-Sensor-SDK #739: Request for depth engine release on debian stretch

ratman, 2019-09-09:
Hi Everyone,
as the title tells, I would like to request Debian packages for genuine debian distros as well, preferably for stretch, but also for buster. Is there hope for that in the near future?

BR.

wes-b, 2019-09-09:
Please head over to  https://aka.ms/azurekinectfeedback to ask for this feature.

closed 2019-09-09 by wes-b

ratman, 2019-09-09:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#debian-package

### "We currently have debian packages available for Ubuntu 18.04. If you have need for a different debian distribution, please file an issue."

Guess where the hyperlink leads:

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/new?assignees=&labels=Enhancement&template=feature-request--enhancement-.md&title=

wes-b, 2019-09-09:
Ahh, thank you for pointing that out. I will update the link