microsoft/Azure-Kinect-Sensor-SDK #741: Creating a 3D Model using extrnal synchronization

JayaSaiSontineni, 2019-09-09:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

JayaSaiSontineni, 2019-09-17:
Hello @tesych ,

This issue is different from #742, This is of more about creating a 3D Model whereas #742 is about 3d 360 volumetric capture video.

Elenaegr, 2020-01-30:
Hi, I am also interested in the question done by @JayaSaiSontineni and I agree that it is not the same than in #742 because they are asking how many cameras are needed/enough. However, we want to know if there is any way to merge the 3D models generated by the different synched kinects into just one 3D model.