microsoft/Azure-Kinect-Sensor-SDK #742: Creating a 3D 360 degree volumetric capture video

JayaSaiSontineni, 2019-09-09:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

tesych, 2019-09-16:
closing this issue, but if you have more questions, please let us know.

closed 2019-09-16 by tesych

Elenaegr, 2020-01-31:
Hello @tesych , I would be interested in knowing if there is any way to merge the 3D models created from different synched kinects into just one. If so, I would like to know the way to do so or any tips to begin with. Thanks!

wes-b, 2020-01-31:
Check out our green screen sample. It demonstrates how to transform depth data from 1 camera coordinate system to another cameras. It also uses OpenCv to calculate the transformation necessary to convert images between the two cameras.

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/green_screen

Elenaegr, 2020-02-18:
Hello @wes-b, 
I tried the green_screen example with the following parameters for both configurations (one camera and two synched cameras) and the result seems to be exactly the same one. See the following screenshots:

Result when running it for two cameras **>green_screen.exe 2 9 6 22**
[image]

Result when running for it one camera **>green_screen.exe 1 9 6 22**
[image]

Which are the differences I should see? I was expecting to get better resolution for the 2 cameras configuration but it doesn't seem to be any. Any help would be appreciated.

wes-b, 2020-02-20:
The demo works best when the camera's are only a few inches apart and the color (master) is between the two depth camera's.

See this: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/green_screen