microsoft/Azure-Kinect-Sensor-SDK #743: Not able to build the package for Body tracking count using sample code provided

JayaSaiSontineni, 2019-09-09:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

qm13, 2019-09-09:
What steps did you go through? What happened?

JayaSaiSontineni, 2019-09-10:
Hi,

I have followed the steps specified in the page https://docs.microsoft.com/en-us/azure/kinect-dk/build-first-body-app.

And then, I tried building the application package using Debug, X-64 mode.

It has popped an alert saying clear all errors before building the package, but no errors were shown at the output list.

Thanks,
Jaya Sai Sontineni
Holo Pundits | Software Developer, Applications
+1 8166168778  | www.holopundits.com<http://www.holopundits.com/>
*** HoloPundits is a Global Technology Company Specializing in developing AR, MR, and VR Applications***

[Logo]
ISO 9001:2015 & ISO 27001:2013 Certified Company

[mixed-reality_badge_side-by-side_blk-blu]
CONFIDENTIALITY NOTICE: This e-mail message, including any attachments, is for the sole use of the intended recipient(s) and may contain confidential and privileged information. Any unauthorized review, use, disclosure or distribution is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

yijiew, 2019-09-16:
@JayaSaiSontineni Did you include all the necessary headers and libs? Instead of creating your own project, you could directly download the sample projects here: https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples

JayaSaiSontineni, 2019-09-16:
Hi,

I just downloaded the sample project from the link you shared.

Regards,
Jaya Sai

yijiew, 2019-09-16:
@JayaSaiSontineni Awesome! Does it work?

JayaSaiSontineni, 2019-09-17:
It's the same issue again!!

yijiew, 2019-09-17:
@JayaSaiSontineni Can you share some screenshot of the errors?

JayaSaiSontineni, 2019-09-17:
Thanks,
Jaya Sai Sontineni
Holo Pundits | Software Developer, Applications
+1 8166168778  | www.holopundits.com<http://www.holopundits.com/>
*** HoloPundits is a Global Technology Company Specializing in developing AR, MR, and VR Applications***

[Logo]
ISO 9001:2015 & ISO 27001:2013 Certified Company

[mixed-reality_badge_side-by-side_blk-blu]
CONFIDENTIALITY NOTICE: This e-mail message, including any attachments, is for the sole use of the intended recipient(s) and may contain confidential and privileged information. Any unauthorized review, use, disclosure or distribution is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

yijiew, 2019-09-17:
I don't see your screenshot pasted... Could you click the GitHub link and paste your screenshot there instead of directly replying to the notification email?

JayaSaiSontineni, 2019-09-17:
[image]

JayaSaiSontineni, 2019-09-17:
When I try to remove and build it in release mode, It is showing an error of missing stdafx.h library file.

yijiew, 2019-09-17:
We don't include the stdafx.h file in any of our samples. Where do you get it from? Please try to remove this file and build it. Since this is not related to the SDK, I will close this issue.

closed 2019-09-17 by yijiew

JayaSaiSontineni, 2019-09-17:
If I remove stdafx.h file from the sample, here below is the screenshot of error that was popping up.
[image]

JayaSaiSontineni, 2019-09-17:
Dear Yijie,

Please reopen the issue as it was not yet addressed by Microsoft team and I am seeking some support from Mcirosoft.

Thanks,
Jaya Sai Sontineni

Closed #743<https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/743>.

yijiew, 2019-09-18:
Friendly tip: you can find the solution by google the error message: "unexpected end of file while looking for precompiled"

Here is the solution: https://stackoverflow.com/questions/20113280/fatal-error-c1010-stdafx-h-in-visual-studio-how-can-this-be-corrected

JayaSaiSontineni, 2019-09-18:
I have tried both ways mentioned in that page earlier to my comment or posting on github .

But still, I was not able to find the drop down section for Configuration.

Regards,
Jaya Sai